rusefi/rusefi #1105: lock-free implementation of getTimeNowNt

rusefi, 2020-01-19:
```#else /* EFI_PROD_CODE */
// todo: why is this implementation not used?
	/**
	 * this method is lock-free and thread-safe, that's because the 'update' method
	 * is atomic with a critical zone requirement.
	 *
	 * http://stackoverflow.com/questions/5162673/how-to-read-two-32bit-counters-as-a-64bit-integer-without-race-condition
	 */
	efitime_t localH;
	efitime_t localH2;
	uint32_t localLow;```

rusefillc, 2021-11-07:
@racer-coder any interest?

racer-coder, 2021-11-07:
yes

racer-coder, 2021-11-07:
BTW I think LDRD would be a cheaper answer here.  If the instruction is interrupted it restarts (unlike LDM which will continue mid-instruction).  No need to loop on load.  It is ARM specific though (M3 or later, I think).   Can always leave the locking version as a fallback for x86, or PIC, or ??

Unfortunately, STRD won't be possible, as (I believe) the partial store succeeds, gets interrupted, then restarted.

Actually even better - if you don't need full 64-bit support, but even just 63-bit support, then you only need to store the top bit of the lower 32-bit bits along with the 31-bits from the top.  That would require only a single 32-bit load or store, as long as the function is called every 1 billion (1<<30) cycles.  Is that a reasonable enough ask?  Actually the existing code has the same problem anyways, except at 2 billion cycles.

mck1117, 2021-11-07:
> Can always leave the locking version as a fallback for x86, or PIC, or ??

I'm perfectly happy to have an armv7m-specific version of this function.

> as long as the function is called every 1 billion (1<<30) cycles. Is that a reasonable enough ask

Well, we currently call it thousands of times a second, so we're probably safe there.

mck1117, 2021-11-07:
We run the timer at 4mhz, so 1 billion cycles is 250 seconds, a little over 4 minutes.

rusefillc, 2021-11-08:
#3502

closed 2021-11-08 by rusefillc