rusefi/rusefi #1120: Support stepper with TLE8888

rusefi, 2020-02-02:
looks like microRusEfi hardware is set to use two H-bridges for stepper control already. Maybe it would be possible for TLE8888 to control stepper IAC?

This task probably consists of TLE8888 initialization part and generic https://github.com/rusefi/rusefi/issues/1088

rusefi, 2020-02-02:
@dron0gus would you be interested to implement the TLE8888 initialization part or the whole change?

mck1117, 2020-02-02:
Hmm, this may actually be zero additional work once #1121 goes in.  The tle8888 outputs just need to get configured as DC motors (which you'll be able to do with #1121, since they're just normal GPIO), so this may just work.

dron0gus, 2020-02-02:
@mck1117 I agree.
We can just hack/modify tle8888 driver, so OUT21..OUT24 will be initialized as Push-Pulls.
Or
We can use ->setPadMode gpio-driver hook to set OD or PP mode. This is not implemented yet, I can do it.
But, in #1121 you are using DcMotor which is using SimplePwm which is initializing OutputPin with simple initPin(msg, brainPin) without special mode... So gpio-driver hook will be called with OM_DEFAULT mode. As it is done for all other pins used in Open-Drain mode.
There is some mess with output pin modes. For example for Injector pins we shouls use OD mode when using external gpio-chip. And push-pull when MOSFET is driven directly from MCU pin.
Also there is some mess with logic levels. We pass '1' to gpio-chip to open MOSFET and actually output '0'. Same question for Push-Pulls output of TLE8888. What value should be set on output when output is PP and tle8888_writePad(data, pin, 1) is called?
Sorry for offtopic.

mck1117, 2020-02-05:
I think for now we could just make those tle8888 outputs permanently pushpull, and come back around to setting them properly later.

re: polarity, for a stepper it actually doesn't matter.  So for this particular use case nobody will notice if it's inverted or not, since the stepper will still work fine.  We should open a separate issue for figuring out what to do about tle8888 (and similar gpiochip) outputs with a configurable mode.

dron0gus, 2020-02-13:
https://github.com/rusefi/rusefi/pull/1125/commits/413839f6cd4339a22761fc55903df3ad76d7ddb0 and https://github.com/rusefi/rusefi/pull/1125/commits/342c62461fb01f19734d6fd26ad10588f14910a4

rusefillc, 2020-11-11:
See https://github.com/rusefi/rusefi/pull/1936
See https://github.com/rusefi/rusefi/pull/1125

rusefillc, 2021-05-16:
JRD McLAREN  4:58 AM
(IDLE stepper coil 2+) are twice on https://rusefi.com/docs/pinouts/microrusefi/ on PIN33 and PIN43 ..
4:58
one of them seems to be "minus" ...

dron0gus, 2022-01-07:
TLE8888 fix https://github.com/rusefi/rusefi/pull/3748
Settings https://github.com/rusefi/rusefi/pull/3749
Code https://github.com/rusefi/rusefi/pull/3751

dron0gus, 2023-05-07:
@rusefillc done?

closed 2023-05-07 by rusefillc