rusefi/rusefi #1167: Another 55 pin footprint and breakout board

rusefi, 2020-03-03:
We have a breakout board for 963063 https://github.com/rusefi/rusefi/tree/master/hardware/963063-15_55_pin_connector

there is a bit different compatible connector with smaller spacing between terminal rows on the PCB and two plastic alignment pins

We have GERBERS for the "other" version (PN unknow)

We need a Kicad module/footprint/component for the "other" version and a breakout board for that "other 55" https://github.com/andreika-git/prometheus/blob/master/hardware/55-pin-connector-adapter-963063-15-gerber.zip

rusefi, 2020-03-03:
See also https://rusefi.com/forum/viewtopic.php?p=35546#p35546

rusefi, 2020-03-03:
Another way to say the same:

We have motronic gerbers, no motronic  kicad. motronic kicad is needed.

looks like 963063 kicad could be a great starting point for motronic connector

i believe PCB footprint row spacing would need to go down and two holes would be needed for pivoting pins

rusefi, 2020-03-04:
[image]

rusefi, 2020-03-04:
here are the non-kicad gerbers

open question what's going on with row spacing - what is footprint row spacing on this PCB? my assumption was that we are dealing with two different spacings but maybe this assumption is wrong?

[yanvar51_gerber.zip](https://github.com/rusefi/rusefi/files/4285607/yanvar51_gerber.zip)

ksmola, 2020-04-10:
Is this done? I built gerbers (and eagle files) of a 55pin Motronic connector. I designed a pcb to use a standalone ECU 'plug and play' in the Motronic case with original connector so I know they fit.

rusefi, 2020-04-10:
this is done https://github.com/rusefi/rusefi/tree/master/hardware/breakout_55_pin

closed 2020-04-10 by rusefi