rusefi/rusefi #1170: DFU does not work with latest windows drivers?

rusefi, 2020-03-05:
See https://rusefi.com/forum/viewtopic.php?p=35602#p35602

rusefi, 2020-03-06:
Looks like ST did not update DfuUtil, same version 3.0.6 is on the website?

* Version            : V3.0.6
* Date               : 01-June-2018

chaoschris, 2020-03-06:
Tested it also with version 3.0.6 direct  from STM:

C:\Program Files (x86)\STMicroelectronics\Software\DfuSe v3.0.6\Sources\Binary\R
elease>dir
 Volume in Laufwerk C: hat keine Bezeichnung.
 Volumeseriennummer: 0081-9799

 Verzeichnis von C:\Program Files (x86)\STMicroelectronics\Software\DfuSe v3.0.6
\Sources\Binary\Release

03.03.2020  16:56    <DIR>          .
03.03.2020  16:56    <DIR>          ..
10.06.2018  23:54            49.664 DfuFileMgr.exe
10.06.2018  22:24            28.160 DfuSeCommand.exe
10.06.2018  23:33         1.925.632 DfuSeDemo.exe
10.06.2018  23:41             4.303 SLA0044.txt
10.06.2018  21:38            72.704 STDFU.dll
10.06.2018  23:33            33.792 STDFUFiles.dll
10.06.2018  23:45            28.672 STDFUPRT.dll
29.09.2012  21:17         1.480.704 STDFUTester.exe
10.06.2018  21:35         1.800.192 STTubeDevice30.dll
               9 Datei(en),      5.423.823 Bytes
               2 Verzeichnis(se), 96.039.772.160 Bytes frei

C:\Program Files (x86)\STMicroelectronics\Software\DfuSe v3.0.6\Sources\Binary\R
elease>DfuSeCommand.exe -c -d --v
0 Device(s) found. Plug your DFU Device !
file does not exist

 Press any key to continue ...

chaoschris, 2020-03-06:
Trying to send the dfu file brings up error 12340005, but i'm sure this has nothing to do with the error as DfuSe didn't find any device to connect to....

C:\Program Files (x86)\STMicroelectronics\Software\DfuSe v3.0.6\Sources\Binary\R
elease>DfuSeCommand.exe -c -d --v --fn rusefi.dfu
0 Device(s) found. Plug your DFU Device !

Target 00: Error Code: Unknown error 0x12340005.
Target: 00
Error Code: Unknown error 0x12340005.
TransferSize: 0
DFU State: (Unknown 0x76)
DFU Status: STATUS_OK
Request: (unknown 0x062A8AB5).
CurrentNBlock: 0xB0000000
CurrentLength: 0x300076FB
Percent: 0

chaoschris, 2020-03-06:
And i forgot to mention, same behaviour on Windows 7 64Bit

rusefi, 2020-03-10:
Let's discuss this at https://app.slack.com/client/T4NJHQ8QZ/CF80PQGMV/details/members

rusefi, 2020-03-10:
Windows 7 stuff works for me today with the following driver:

[image]

mck1117, 2020-03-10:
Win10 does NOT work for me, shows this as the driver:
[image]

Device vid/pid is 0483:df11, so that's certainly the correct device.  STM32CubeProgrammer can connect to it properly, but the rusEfi console can't, failing with the following error:

```
Giving time for USB enumeration...
Executing .\.\DfuSe/DfuSeCommand.exe -c -d --v --fn ..\rusefi.dfu
0 Device(s) found. Plug your DFU Device ! 

Target 00: Error Code: Unknown error 0x12340005. 
Target: 00
Error Code: Unknown error 0x12340005. 
TransferSize: 0
DFU State: STATE_IDLE
DFU Status: STATUS_OK
Request: (unknown 0x1C222AA3). 
CurrentNBlock: 0x1975052D
CurrentLength: 0x75FDEB
Percent: 0
```

Note that it did successfully connect to the firmware and reboot it to DFU mode, but couldn't connect.

rusefi, 2020-03-11:
I was trying to reproduce the issue on my Windows 10:

1) step one - launch rusEfi console, hit "Auto DFU"
2) Device manager shows "stm32 boot-loader" non functioning device, drivers did NOT miraculously appear
2) I hit "update drivers", "automatic"

and I got the (good?) driver installed - maybe I've played with rusEfi driver bundle here before?
[image]

and at the end I have NOT working DFU like this:

[image]

which could be another issue not the original issue reported here?

rusefi, 2020-03-11:
https://wiki.st.com/stm32mpu/wiki/STM32CubeProgrammer mentions uninstalling our current DfuSe driver. 

Looks like STM32CubeProgrammer is the new way for stm32 DFU.

Looks like one day we would need to migrate to this new way.

rusefillc, 2022-11-27:
We've migrated to STM32 BOOTLOADER driver

closed 2022-11-27 by rusefillc