rusefi/rusefi #1171: CLT stops working when using 2-0 custom trigger wheel

ZHoob2004, 2020-03-06:
MicroRusefi 0.1 hardware
snapshot_20200302_141558_mre-f7_rusefi

Steps to reproduce:

1 - Erase STM32 using STM32CubeProgrammer
2 - Program bundle firmware
3 - Create new TunerStudio project using matching .ini
4 - Change trigger settings to custom, set teeth to 2 and skipped teeth to 0
5 - Write and reboot ECU to apply settings

CLT sensor exceeds max value and displays 70c, forcing radiator fan on

rusefi, 2020-03-06:
Does it show FATAL?

Also please try this:
1 - Erase STM32 using STM32CubeProgrammer
2 - Program bundle firmware
3 - Create new TunerStudio project using matching .ini
4 - Write and reboot ECU to apply settings

what would it show WITHOUT changing trigger?

in rusEfi console please execute
_tempinfo_
command and attach outpur

ZHoob2004, 2020-03-06:
Yes fatal error.

CLT works as expected without setting custom trigger. If I change back to 60/2 the CLT starts working again. Setting back to 2-0 breaks it again.

tempinfo
```
2020-03-05_20_08_17_428: EngineState: confirmation_tempinfo:8
2020-03-05_20_08_17_430: EngineState: CLT volts=0.00 Celsius=0.00 sensorR=0.00 on channel 0
2020-03-05_20_08_17_431: EngineState: @PA0
2020-03-05_20_08_17_432: EngineState: C=-20.00/R=12000.00 C=0.00/R=5000.00 C=100.00/R=200.00
2020-03-05_20_08_17_432: EngineState: bias resistor=2.70K A=0.00123 B=0.00026 C=0.00000
2020-03-05_20_08_17_434: EngineState: ==============================
2020-03-05_20_08_17_435: EngineState: IAT volts=0.00 Celsius=0.00 sensorR=0.00 on channel 1
2020-03-05_20_08_17_436: EngineState: @PA1
2020-03-05_20_08_17_436: EngineState: C=-20.00/R=12000.00 C=0.00/R=5000.00 C=100.00/R=200.00
2020-03-05_20_08_17_437: EngineState: bias resistor=2.70K A=0.00123 B=0.00026 C=0.00000
2020-03-05_20_08_17_438: EngineState: ==============================
2020-03-05_20_08_17_440: EngineState: fan=Yes @ ext:(null).-1
2020-03-05_20_08_17_440: EngineState: A/C relay=Yes @ NONE
```

Also looks noteworthy:
`2020-03-05_20_08_17_346: EngineState: FATAL error: trigger constraint violation`

closed 2020-03-07 by rusefi