rusefi/rusefi #1181: rusEfi console does not start on Windows 10

rusefi, 2020-03-10:
I am sure I have java 8 installed yet rusEfi console launcher says it requires version 1.7

console definitely should run under Azul or any other java 8

rusefi, 2020-03-10:
working on following windows 7:

[image]

mck1117, 2020-03-10:
On Windows 10 build 18362 I show:
```
PS C:\Users\matth> java -version
java version "1.8.0_221"
Java(TM) SE Runtime Environment (build 1.8.0_221-b11)
Java HotSpot(TM) 64-Bit Server VM (build 25.221-b11, mixed mode)
```

rusefi, 2020-03-10:
Does not start under

[image]

rusefi, 2020-03-10:
On my Windows 10 does not start under 

[image]

mck1117, 2020-03-10:
installed at
```
PS C:\Users\matth> (get-command java).Source
C:\Program Files (x86)\Common Files\Oracle\Java\javapath\java.exe
```

rusefi, 2020-03-11:
Works on my Windows 10 under another java

[image]

mi-hol, 2022-11-06:
runs with AdoptOpenJDK using PowerShell 7 as command shell using below command
`."C:\Program Files\jdk8u345-b01\jre\bin\java.exe" -jar "C:\Program Files\snapshot_20220914_125452_mre_f4_rusefi\console\rusefi_console.jar"`

Unfortunately the misleading "Java Runtime Environment 1.7.0" information is still displayed.
[image]

Java 7 is only available from ORACLE with a personal account!

I searched with 'site:github.com/rusefi/rusefi "Java Runtime Environment 1.7.0"' and other strings but had no hit unfortunately.
Could the text & code be updated to reference Java 8?

rusefillc, 2022-11-06:
@mi-hol see https://github.com/rusefi/rusefi/tree/master/misc/console_launcher we use an external tool to create those .exe files

I've just used fresh version of the tool maybe it would help please test with latest snapshot once it builds in 30 minutes?

mi-hol, 2022-11-06:
@rusefillc many thanks for this quick response :)

[image]

due to "unknown publisher" but then displays dependency on Java 8
[image]

Shall I open a new issue for  "unknown publisher"?

rusefillc, 2022-11-06:
I assume that "unknown published" is a pre-existing condition

And it sounds like java version check was not fixed by http://launch4j.sourceforge.net/ people. Please help by poking http://launch4j.sourceforge.net/ people directly and linking http://launch4j.sourceforge.net/ here?

mi-hol, 2022-11-06:
I didn't find a wiki page for the simulator in special use-cases cases like "explain running with a specific JAVA version in case multiple JAVA versions are installed".
Theredore my second suggestion would be a wiki page to explain running with several use-cases. 
Would that make sense?

mi-hol, 2022-11-06:
> I assume that "unknown published" is a pre-existing condition

very likely true

> And it sounds like java version check was not fixed by http://launch4j.sourceforge.net/ people.

not sure to understand this comment. From my view it was fixed but these days JAVA 11 or even later version are defaults /as on my PC).

rusefillc, 2022-11-06:
@mi-hol I've lost you big time. I am not aware of any relationship between rusefi_simulator.exe and java.

rusefillc, 2022-11-06:
@mi-hol this ticket is about rusefi_console.exe "rusEFI console launcher" let's keep it focus on only rusefi_console.exe

mi-hol, 2022-11-06:
> I've lost you big time. I am not aware of any relationship between rusefi_simulator.exe and java.

The initial issue "rusEfi console does not start under AdoptOpenJDK" seems solved

Apologize, my fault, I got confused by below message :(
[image]

rusefillc, 2022-11-06:
Closing this one, looking forward any new separate reports!

closed 2022-11-06 by rusefillc