rusefi/rusefi #1204: TPS calibration usability went down

rusefi, 2020-03-21:
TPS calibration usability went down with new sensor implementation

previously we were evaluating settings on each access, now we wait for Burn command and this dialog does not have a Burn button :(

I believe we have to revert back to behavior without A&B pre-calculation and just interpolate on each invocation

mck1117, 2020-03-22:
Just did some testing - this wasn't a regression in our firmware.  TS doesn't send the burn command, so we never copy the value from the working copy to the in-use one.  Either TS stopped sending the burn command (possible), or it never actually worked without pressing the burn button (also possible).

mck1117, 2020-03-22:
Hoooold on a sec - I just got TS to burn by pressing the "accept" button on the TPS dialog.  This is clearly a TS bug, not a bug in our firmware.

mck1117, 2020-03-22:
It seems like it's a dice roll whether it works or not.  I tried flipping back and forth between two values with the exact same sequence of mouse clicks/key presses, and it burned maybe half the time.

Confirmed by showing gauges with both the new and old TPS calculations, and a breakpoint in the `reconfigureTps()` function, that was only hit some of the times the "Accept" button was pressed.

mck1117, 2020-03-22:
Here's the behavior.  The first two worked as they should, but the third never burned after changing the setting.  Note how the burn button turns red when accept is clicked, and the new settings applied after burn is pressed.

https://www.youtube.com/watch?v=wGzJsUrX36A

OrchardPerformance, 2020-05-22:
Can confirm this is present but it seems to be a TS bug. 
Do we close as this is not a rusEFI problem and send a bug report to TS?

rusefi, 2020-05-22:
It would have to start open to keep state of TS bug and would be closed once we have a resolution from TS or else?

We can maybe consider hiding OEM TS dialog altogether for example - we have options?

mck1117, 2021-05-31:
This is fixed as of TS 3.1.04 or so.

closed 2021-05-31 by mck1117