rusefi/rusefi #1240: ETB dual signal implementation

OrchardPerformance, 2020-03-31:
Issue raised to house the outcome of discussion on slack. Aim is to improve ETB implementation and implement basic safety checking. 

- Both dual pedal and dual TPS signals are calibrated as normal
- Both signals are translated and compared for matching pedal(APP)/TPS percentage
- Configurable threshold for the deviation allowed between signal A and B, if this threshold is broken then trigger an error
- Current high and low thresholds will apply to both signals should they go open or short circuit
- If both signals are too similar, i.e. both 0-5v, then trigger an error or disallow the calibration to prevent single signal hacks 
- Single signal mode can be activated for testing or dev work only on a debug build and by setting both a bit in the INI and activating a bench test button to prevent accidental single signal use

rusefi, 2020-04-20:
See also #1336

OrchardPerformance, 2020-05-22:
Can this one now be closed as we have dual signal as a ticket open for the deviation check? 
Seems no need for this one anymore? 

@mck1117 - Do you think this one needs to stay open for the "similar signal check", do we care about that?

OrchardPerformance, 2020-06-01:
As Matt has said he has driven on dual signal I think we can close this unless any problems pop up.

closed 2020-06-01 by OrchardPerformance