rusefi/rusefi #1265: ambiguousOperationMode tooltip 180, 360, or 720

rusefi, 2020-04-05:
@mck1117 

It's about ambiguousOperationMode..

Andrey  11:33 AM
so please explain your proposal about ambiguousOperationMode in English, from start to end

960  11:35 AM
This is the one everyone are asking about and dont understand..
It's about if it needs channel 2 for sync?

Gwendal Blot  11:36 AM
Oh yes like the Renix for example

Matthew Kennedy  11:36 AM
It's about whether the trigger pattern is 180, 360, or 720 degrees long

Andrey  11:37 AM
this "180, 360, or 720" could be a useful comment!
11:37
Let's do a tooltip with this?
11:37
@Matthew Kennedy can you write the complete text?

rusefi, 2020-04-05:
Gwendal Blot  11:41 AM
So you basically have a number of pattern per crankshaft turn, it may be clearer for people

11:41
1, 2,3 or 0.5 for cam

rusefillc, 2022-12-11:
@mck1117 does this seem done to you?

mck1117, 2023-01-20:
ambiguous operation mode is ggggggone!

closed 2023-01-20 by mck1117