rusefi/rusefi #1299: Updates to fuel algorithms

OrchardPerformance, 2020-04-15:
Creating tickets to track the improvements to the fuel algorithms 

1- Fix LM_REAL_MAF 
This means:
a) Modifying firmware so airChargeLoad is used as the value for "engine load" in TS
b) Checking the Y axis of the fuel table uses this value 
c) Checking the Ignition table axis uses this value
d) I will put some miles on it and grab some logs so we know its sorted.

2- Deprecating LM_PLAIN_MAF, 
Not sure what this needs, after discussion with Matt there is no need for it to exist once the tasks above are done. 
Belief is that at present no users are actually using this? 

3- Move AlphaN to VE fuel table.
This is a change to move from a ms pulse width fuel table to a VE table like speed density. This is how other systems work with their AlphaN mode so is a usability improvement.

OrchardPerformance, 2020-05-22:
125JDavis said - Alpha-N fueling algorithm doesn't have VE table or target AFR table. Currently the user tunes by changing target PW values in the TPS/RPM table. Useability would be much better if the same VE/AFR fueling tables from speed density were carried over to Alpha-N.
Updated equation for O2 mass = TPS * Baro Pressure * Displacement / (Gas Constant * Intake Temp)
[image]

mck1117 said - Instead of using TPS directly, I think you want a "VE" value looked up from a table as a function of TPS and RPM. So essentially the same as the speed density math, but with pressure removed, and VE looked up using TPS instead of MAP.

OrchardPerformance, 2020-05-22:
I also think this ticket needs bumping up the order as we have PNP boards needing the MAF and more people looking at AlphaN

OrchardPerformance, 2020-07-28:
Roadmap linked above. 

Some comments on Ethanol flex fuel implementation here for #540 

Density is currently used only in:
constexpr float convertToGramsPerSecond(float ccPerMinute) {
	float ccPerSecond = ccPerMinute / 60;
	return ccPerSecond * 0.72f;	// 0.72g/cc fuel density }

Suggest this is broken out into a "fuel_density.cpp" where we can modify this by ethanol percentage and temperature. 

If we primarily use an ethanol % modifier then we can either have that as a static value or an input from the flex fuel sensor. 
We should probably make this over-ridable for those with known racing fuel or methanol. 

Coefficient of thermal expansion for the fuels "Gasoline is 0.00095 Eth 0.00109 and Meth 0.00149"

Comments from Slack:
"AFR as an eth% input simply as that is one less related to the physical properties and is linear.
Density correction as a linear interpolation between the 0 and 100% eth values?
Configurable or linear interpolated density?
Is the answer yes and allow a tick box override for people with exact fuel blends like methanol?
Is the answer to just default for Gasoline-Ethanol % mix and allow override to account for Meth or race fuel?"

mck1117, 2021-03-15:
@OrchardPerformance I think this one is probably done - looks like all the TODOs here are complete

closed 2021-05-31 by mck1117