rusefi/rusefi #1312: ETB tunerstudio 3D map scaling

125jdavis, 2020-04-18:
1. All values in ETB map are set to -2147483648 by default. 
2. 3D map scaling is odd, doesn't show shape of map because limits are too large

[image]
[image]

* [ Not using dardware]

rusefi, 2020-04-18:
This is odd. What do you mean "not using hardware"? While tunerstudio has a feature for defaults without hardware we rely on hardware or similar to provide defaults. How is simulator working for you?

125jdavis, 2020-04-18:
I don't have any hardware, I just loaded the Frankenso .ini but every table, curve, and scalar is empty. 
I haven't connected to the simulator yet, still trying to figure out how all this stuff works.

rusefi, 2020-04-18:
While technically we can make an extra effort to provide defaults into .ini we also kind of do not bother since high effort low benefit

mck1117, 2020-04-18:
Good catch on #2 - the limits are indeed wrong.  I'll fix that!

mck1117, 2020-04-18:
Units are wrong too.

mck1117, 2020-04-18:
That looks better!

[image]

PR #1316 (2020-04-18): ETB pedal map scale
Was scaled -720 to +720, now scaled 0 to 100.

Fixes #1312
closed 2020-04-18 by rusefi