rusefi/rusefi #1336: ETB deviation detection and response

rusefi, 2020-04-20:
We shall detect if ETB suggestively decides to not behave as commanded and respond to that somehow.

rusefi, 2020-04-20:
Part of the response could be turning off DIS signal?

rusefi, 2020-04-21:
@960 suggest
```float etbError = absF(targetPosition - actualThrottlePosition.Value); 
         if (etbError < 3 ) {
           engine->deviationRpmLimit = false;
           errorTime = getTimeNowNt();
      /All ok, do nothing 
         }
       else {
         }        // ETB Deviation outside limit detected, set timestamp and monitor it for two seconds
         if ((getTimeNowNt() - errorTime > MS2NT(2000)) && (rpm > engineConfiguration->cranking.rpm)) {
        	 engine->deviationRpmLimit = true;
        	 warning(Electronic_Throttle_Position_Control_Error, "Electronic throttle deviation limits exceeded");
    	}
// Todo: Add counter who shows warning the first time, and error after 5 times
```

OrchardPerformance, 2020-05-22:
@mck1117  - Now you have a dual signal working is this one low hanging fruit?

mck1117, 2020-05-23:
perhaps - we should come up with something clever to trip on actual errors, but not trip on transients

OrchardPerformance, 2020-05-23:
Maybe, The way this is handled in a lot of OEM application is to have a counter represent the deviation. The deviation value is added to this counter every XXms so the larger the deviation the faster it hits the error value. 
Question is how we decide to reset the counter? 
When the TPP and TPS match? 
After X number of seconds?

mck1117, 2020-05-23:
Perhaps we add the deviation to some counter (integrating the error), and then subtract from the counter at some constant rate.  So that perhaps we tolerate a 2% error indefinitely, but a 20% error for only 1/10 second.

OrchardPerformance, 2020-05-23:
I see no issue with that, just a TS value for "ETB error tolerance" with a configurable arbitrary value. 
Higher = more tolerance as it subtracts more. 
Set a reasonable value as standard or maybe set it via the autocalibrate? 
Could we detect error during auto cal and decide the suitable baseline tolerance value from that?

mck1117, 2021-03-08:
Looking at some logs, it looks like the gap between "error integral during normal operation" and "error integral during actual failure" is enormous, so you can be pretty loose on exactly what you set the error threshold to.

rusefillc, 2022-11-27:
@mck1117 please blink if there is hope

mck1117, 2022-11-27:
Yes, this and #4833 are good ones for me to tackle this winter.

rusefillc, 2025-04-15:
ErrorAccumulator seems to be unused as of today @dron0gus

rusefillc, 2025-04-15:
Some stuff was removed while adding jam detector https://github.com/rusefi/rusefi/commit/3ee894594d405ceac8a98ea5838c3d11c94e9c1d

fun fact - we do not understand jam detector or maybe it's broken?