rusefi/rusefi #1353: Race condition in OutputPin::setValue

mck1117, 2020-04-23:
OutputPin::setValue uses `palWritePad`.

`palWritePad` is implemented by reading the port, permuting bits, and then writing the port to set the pin to the desired state.

This operation is not atomic - if a context switch occurs inbetween reading the previous state and writing the state and it sets an output pin on the same port, it's possible for the IO port to get in to an incorrect state.

See the warning in the header near `palWritePad`:

```
* @note    The operation is not guaranteed to be atomic on all the
 *          architectures, for atomicity and/or portability reasons you may
 *          need to enclose port I/O operations between @p osalSysLock() and
 *          @p osalSysUnlock().
```

On the stm32, the macros `palSetPad` and `palClearPad` ARE indeed atomic, since they write to the `BSRR` registers (single instruction to set/clear a bit in the `ODR`).

mck1117, 2020-04-23:
I have no evidence that this has actually happened - the window for an interrupt to happen between the read/write is at most 3-5 cycles long, so it would require some incredible luck.  However, the effects could be pretty bad: firing a coil early (or late), closing an injector early or late, etc.

dron0gus, 2020-05-01:
Looks like GPIOv2 LLD is used for STM32F4xx:
```
#define pal_lld_writegroup(port, mask, offset, bits)                        \
  ((port)->BSRR.W = ((~(bits) & (mask)) << (16U + (offset))) |              \
                     (((bits) & (mask)) << (offset)))
```

mck1117, 2020-05-01:
Oh, I'm not sure how I missed that. I thought the implementation I'd seen was read-modify-write, but evidently not.

closed 2020-05-01 by mck1117