rusefi/rusefi #1369: Auto generate requiresPowerCycle for all IO pins

mck1117, 2020-04-26:
I've seen users get tricked a few times when they expected inputs to be able to switch with the ECU running.  TS has the feature, we just need to generate lines in the ini for all input/output pins to make TS show a warning when you change one.

rusefi, 2020-04-26:
I believe the strategy is to respond to as many changes as possible on the fly. We already can change ignition and injection pins on the fly

At the moment we have a pattern of conditional subsystem initialization on start-up only I guess that should be changed so that subsystems could be turned on on the fly.

mck1117, 2020-04-26:
While technically possible to support any reconfiguration possible without a reboot, I'm not sure it's worth it, or even a good idea.  What do we do about systems that require multiples pins to be set, when a partially applied configuration could be invalid or dangerous?  What if you need to set up the pedal and both TPS sensors, but the ETB decides that it's alive after just configuring the pedal?

mck1117, 2020-04-26:
not to mention the logic to try and synchronize that can get a bit hairy.  I've seen cases where the trigger decoder got stuck in half RPM mode, even though it was maintaining sync and the config indicated it was in crank speed mode.

mck1117, 2021-03-19:
duplicate of #1231

closed 2021-05-06 by rusefillc