rusefi/rusefi #1394: StepperMotor casue system halt on configuration change

dron0gus, 2020-05-03:
It tries to start already running thread:

0x08001cba in chSysHalt (reason=0x8066110 "working area in use") at ChibiOS/os/rt/src/chsys.c:209
209	  CH_CFG_SYSTEM_HALT_HOOK(reason);
(gdb) bt
#0  0x08001cba in chSysHalt (reason=0x8066110 "working area in use") at ChibiOS/os/rt/src/chsys.c:209
#1  0x08002bde in chThdCreateStatic (wsp=0x200152d8 <iacMotor+8>, size=816, prio=128, pf=0x803a611 <ThreadController<400>::StaticThreadTaskAdapter(void*)>, arg=0x200152d0 <iacMotor>)
    at ChibiOS/os/rt/src/chthreads.c:335
#2  0x0803a5e0 in ThreadController<400>::Start (this=0x200152d0 <iacMotor>) at ./controllers/system/thread_controller.h:60
#3  0x0803a492 in StepperMotor::initialize (this=0x200152d0 <iacMotor>, hardware=0x200152bc <iacHbridgeHw>, totalSteps=150, sharedLogger=0x0) at ./hw_layer/stepper.cpp:167
#4  0x08049472 in initIdleHardware () at ./controllers/actuators/idle_thread.cpp:627
#5  0x08035480 in applyNewHardwareSettings () at ./hw_layer/hardware.cpp:374
#6  0x080276a4 in incrementGlobalConfigurationVersion () at ./controllers/algo/engine_configuration.cpp:184
#7  0x0804cb8a in setEngineType (value=30) at ./controllers/settings.cpp:318
#8  0x0804eb32 in setValue (paramStr=0x2001e9fc <handleBuffer+4> "engine_type", valueStr=0x2001ea08 <handleBuffer+16> "30") at ./controllers/settings.cpp:1308
#9  0x08058fe2 in handleActionWithParameter (current=0x1000e0c8 <consoleActions+1456>, parameter=0x2001ea08 <handleBuffer+16> "30") at ./util/cli_registry.cpp:267
#10 0x08059742 in handleConsoleLineInternal (commandLine=0x200127a4 <tsChannel+12> "set engine_type 30", lineLength=18) at ./util/cli_registry.cpp:546
#11 0x08059804 in handleConsoleLine (line=0x200127a4 <tsChannel+12> "set engine_type 30") at ./util/cli_registry.cpp:589
#12 0x0803f230 in handleExecuteCommand (tsChannel=0x20012798 <tsChannel>, data=0x200127a4 <tsChannel+12> "set engine_type 30", incomingPacketSize=18) at ./console/binary/tunerstudio.cpp:697
#13 0x0803f464 in tunerStudioHandleCrcCommand (tsChannel=0x20012798 <tsChannel>, data=0x200127a4 <tsChannel+12> "set engine_type 30", incomingPacketSize=19) at ./console/binary/tunerstudio.cpp:759
#14 0x0803ecc0 in runBinaryProtocolLoop (tsChannel=0x20012798 <tsChannel>) at ./console/binary/tunerstudio.cpp:582
#15 0x0803ed38 in tsThreadEntryPoint (arg=0x0) at ./console/binary/tunerstudio.cpp:595
#16 0x08000342 in _port_thread_start () at ChibiOS/os/common/ports/ARMCMx/compilers/GCC/chcoreasm_v7m.S:122

rusefi, 2020-05-04:
@mck1117 ?

mck1117, 2020-05-04:
BTW, many thanks for including a stack trace with the bug report

mck1117, 2020-05-04:
I'll have a look at a fix in the morning

PR #1396 (2020-05-05): don't fail on multiple start calls
fix #1394
closed 2020-05-05 by rusefi