rusefi/rusefi #1395: HW or SW: TTL on Frankenso is not well

rusefi, 2020-05-04:
My primary communication channel with rusEFI is USB and it seems to be working fine.

For Bluetooth we need TTL and TTL seems to not be well on Frankeno or MRE. Open question if HW or SW.

rusefi, 2020-05-04:
Index: firmware/config/stm32f4ems/efifeatures.h
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
--- firmware/config/stm32f4ems/efifeatures.h	(revision 2337a89cf9a215cd8b8a512c57d044d6e92f560f)
+++ firmware/config/stm32f4ems/efifeatures.h	(date 1588613963958)
@@ -327,9 +327,11 @@
 
 // todo: start using consoleUartDevice? Not sure
 #ifndef EFI_CONSOLE_SERIAL_DEVICE
-#define EFI_CONSOLE_SERIAL_DEVICE (&SD3)
+//#define EFI_CONSOLE_SERIAL_DEVICE (&SD3)
 #endif
 
+#define EFI_CONSOLE_UART_DEVICE (&UARTD4)
+
 /**
  * Use 'HAL_USE_UART' DMA-mode driver instead of 'HAL_USE_SERIAL'
  *
@@ -338,16 +340,22 @@
  *  STM32_UART_USE_USARTx
  * in mcuconf.h
  */
-#define TS_UART_DMA_MODE FALSE
+#define TS_UART_DMA_MODE TRUE
 
 #define TS_UART_DEVICE (&UARTD3)
-#define TS_SERIAL_DEVICE (&SD3)
+//#define TS_SERIAL_DEVICE (&SD3)
 
 // todo: add DMA-mode for Console?
 #if (TS_UART_DMA_MODE || TS_UART_MODE)
 #undef EFI_CONSOLE_SERIAL_DEVICE
 #endif
 
+#undef EFI_CONSOLE_TX_BRAIN_PIN
+#define EFI_CONSOLE_TX_BRAIN_PIN GPIOA_0
+
+#undef EFI_CONSOLE_RX_BRAIN_PIN
+#define EFI_CONSOLE_RX_BRAIN_PIN GPIOA_1
+
 // todo: start using consoleSerialTxPin? Not sure
 #ifndef EFI_CONSOLE_TX_BRAIN_PIN
 #define EFI_CONSOLE_TX_BRAIN_PIN GPIOC_10

rusefi, 2020-05-04:
[image]

rusefi, 2020-05-04:
[image]

rusefi, 2020-05-15:
@mck1117 see https://github.com/rusefi/rusefi/tree/ttl_issues_1395

on that branch I am trying UART DMA. rusEFI does not have complete flexibility in terms which port could run with what settings, so I have to change settings to disable USB connectivity and have two TTL connectors instead. PC10/PC11 with DMA mode runs as SECONDARY connector for... historical reasons

rusefi, 2020-05-15:
@mck1117  says my cable could require 5v while we use 3v

https://files.slack.com/files-pri/T4NJHQ8QZ-F0142AMNRC1/image_from_ios.jpg

rusefi, 2020-06-14:
Alex Miculescu  says " i keep loosing connection to the board on my frankenso, and i do not know if it is soldering or the issue of DMA that you sad.
i only have FT232 connection.... usb not available, just serial connection, it works then i have random disconnects reconnects, then stable for a while

i have no idea where to start the debug of the connection issue"

shadowm60, 2020-06-14:
I have added a Pull up resistor of 10K on the Tx line from the controller to overcome the 3.5 v limit that FT chip expects.

If i enable tx pin to be OD, TS will not even connect to the board.... that is somehow normal? should it be inverted if it is OD? I'll test that theory as well, but with this mod at least i see correct levels without SW change

[https://rusefi.slack.com/files/US6GD434Y/F015D864D5G/image.png](url)

rusefi, 2020-06-14:
[image]

shadowm60, 2020-06-15:
Tested TS connection with STM32F4 discovery and ttl2usb that is working on 3v3, the issue is there, so i tend to say it is a sw issue

rusefi, 2020-06-15:
@shadowm60 above there is info on how to compile custom version with DMA driver, can you give it a try?

shadowm60, 2020-06-15:
I applied the patch manually, and none of the connections are working, cannot connect nor with TS or serial, used PA0 and PA1 for console, removed PA0 from config so i would not get config errors, but no communication could be established. i have to set up a working debugger to be able to get more info.

shadowm60, 2020-06-15:
DMA enabled, i managed to debug till this point: When using TS communication Settings, i hit Test Port -> this will lead to tunerStudioError("ERROR: CRC header size");

shadowm60, 2020-06-16:
I started again to test the DMA patch fb27f89

Now i can connect to TS, so probably i missed something, but the issue is there, so UART or DMA is the same result.

shadowm60, 2020-06-16:
Switched to TS_UART_MODE FALSE, now i can communicate with the board fine. Is there a reason why we should user UART instead of Serial SD3?

rusefi, 2020-06-16:
if the issues a SW 
``` * rusEFI can communicate with external universe via native USB or some sort of TTL mode
 * We have an interesting situation with TTL communication channels, we have
 * 1) SERIAL - this one was implemented first simply because the code was readily available (works on stm32)
 *    this one is most suitable for streaming HAL API
 *    this one is not great since each byte requires an IRQ and with enough IRQ delay we have a risk of data loss
 * 2) UART DMA - the best one since FIFO buffer reduces data loss (works on stm32)
 *    We have two halves of DMA buffer - one is used for TTL while rusEFI prepares next batch of data in the other side.
 *    We need idle support in order to not wait for the complete buffer to get full in order to recieve a message.
 *    Back when we were implementing this STM32_DMA_CR_HTIE was not available in ChibiOS driver so we have added it.
 *    we have custom rusEFI changes to ChibiOS HAL driver v1
 *    F7 uses driver v2 which currently does not have rusEFI changes.
 *    open question if fresh ChibiOS is better in this regard.
 * 3) UART this one is useful on platforms with hardware FIFO buffer like Kinetis.
 *    stm32 does not have such buffer so for stm32 UART without DMA has no advantages
```

rusefi, 2020-06-16:
Alex Miculescu  4:41 PM
ok, dma is working
4:42
PC10, 11 used for TS configured with DMA, i use on the same pins console if needed, i can run both on FT232 conenction on my frankenso
4:42
i'll push the code, and then we can figure out what's next

Andrey  4:47 PM
hold on
4:47
what code push where?
4:47
what did we figure out?
4:47
do you have same hardware and same test
4:47
where DMA works better?
New

Alex Miculescu  4:57 PM
i got to the point that my frankenso is usable, uart with DMA, if i get a release and flash the device, over tll i cannot do anything, it will keep disconnecting if a table is edited, with this config frankenso is usable over ttl, the problem is that we loose usb serial port, i do not mind since i do not have this :)))
4:58
i was thinking this is the target, but if load should be measured, i think that is the next step

Andrey  4:58 PM
please use simpler terms

Alex Miculescu  4:58 PM
sorry its late and I had a long day

Andrey  4:58 PM
"do you have same hardware and same test. where DMA works better?"

Alex Miculescu  4:58 PM
so

Andrey  4:58 PM
YES/NO?

Alex Miculescu  4:59 PM
YES
4:59
but!

Andrey  4:59 PM
GOOD :slightly_smiling_face:
4:59
SHIT :slightly_smiling_face:

Alex Miculescu  4:59 PM
not as you would like to see performance comparison
4:59
i have a board
4:59
no serial usb on it only ttl
4:59
if i download a release
5:00
it is working - i can create a project in TS, but it will disconnect and it is not stable
5:00
with my uart dma, it is stable, i edited tables, sent configurations, and did not crashed
5:00
so from my point of view is ok

Matthew Kennedy  5:00 PM
ah, instability was about current master
5:00
DMA is more stable
5:00
cool :slightly_smiling_face:

Alex Miculescu  5:00 PM
Now what you would like is comparison serial vs uart dma

Andrey  5:00 PM
OK, so you have some test where DMA is more stable. progress.
5:01
I do not follow :disappointed:

Alex Miculescu  5:01 PM
what would you like

Andrey  5:01 PM
"serial usb" is a red flag to me (edited) 
5:01
it's "just" USB
5:01
it's serial TTL vs DMA TTL
5:01
both things FT232
5:01
but different software working with same FT232

Alex Miculescu  5:02 PM
Yes, this is what i understand that you would like now: performance test between serial and uart dma

Andrey  5:02 PM
"performance" is a vague term
5:02
I would say "stability" comparison
5:03
performance hints at speed while the concurn is mostly about stability
5:03
thank you for your patience :slightly_smiling_face:

Alex Miculescu  5:04 PM
now i look at the patch and get confused.... i think i have an idea how to present this/test/explain
5:04
please give me a bit more time, today i am not able to do more work, but i prepare some inputs
5:04
so you can take a look
5:05
I'll keep you posted

Andrey  5:06 PM
:slightly_smiling_face: sweet

shadowm60, 2020-07-02:
So at the end late response, i did a fresh build and started from scratch  to understand how it is all connected and found out that it is not working. DMA cannot be used for UART since it is taken by SPI.

#define STM32_SPI_SPI2_RX_DMA_STREAM        STM32_DMA_STREAM_ID(1, 3)
#define STM32_SPI_SPI2_TX_DMA_STREAM        STM32_DMA_STREAM_ID(1, 4)

#define STM32_UART_USART3_RX_DMA_STREAM     STM32_DMA_STREAM_ID(1, 1)
#define STM32_UART_USART3_TX_DMA_STREAM     STM32_DMA_STREAM_ID(1, 3)---SPI2
#define STM32_UART_UART4_RX_DMA_STREAM      STM32_DMA_STREAM_ID(1, 2)
#define STM32_UART_UART4_TX_DMA_STREAM      STM32_DMA_STREAM_ID(1, 4)---SPI2

BAsically current SPI will take the DMA stream and it will not be there for uart.

There could be a way, rewrite spi driver and init SPI 2 without DMA support, only CJ125 is there for frankenso, but i do not know if the price is really affordable.

rusefi, 2020-07-02:
@shadowm60 I think I have switched master to use DMA for Frankenso and disabled HIP9011 for now :( i was supposed to test BT but never got to it. is your TTL reliable with master now by any chance?

one day in the future I would like to understand this DMA conflict a bit better

rusefi, 2020-07-19:
looks like with UART DMA things are much better now see https://rusefi.com/forum/viewtopic.php?p=37479#p37479

but new DMA buffer shortage conflict :(

closed 2020-07-19 by rusefi