rusefi/rusefi #1415: firmware date in git - automation

rusefi, 2020-05-10:
we need to update firmware build date automatically - at the moment 'getRusEfiVersion' hard-coded date is modified manually

rusefi, 2020-06-16:
@chuckwagoncomputing this one is about changing bottom of engine_controller.cpp on days when we had commits

open question is how can GH action itself push into git

chuckwagoncomputing, 2020-06-16:
I think something called a git hook would be better for that but I've never used one before so I'll have to look into it.

rusefi, 2020-06-16:
Yes, git hook sounds very relevant! No idea if GH supports those.

chuckwagoncomputing, 2020-06-16:
Looks doable, the downside would be that engine_controller.cpp would be modified with every commit.
Another option would be to utilize the preprocessor.
https://stackoverflow.com/questions/11697820/how-to-use-date-and-time-predefined-macros-in-as-two-integers-then-stri

rusefi, 2020-06-16:
preprocessor is a good 90% solution

the use-case where processor does not help is users compiling their own old version of the firmware. while this use-case is tiny, anything in git gives just a bit more certainty

we also have svnversion.h which is somewhat similar - and it even has git hash. those are just too hard to read so at the moment the only usecase for svnversion.h is pushing a refresh of it in order to trigger all the automation. As a human I do not benefit from GIT_HASH and VCS_VERSION

chuckwagoncomputing, 2020-06-17:
Looks like it is possible for a github action to push a change.
Maybe it could be part of the future bundle action, and use the date the bundle was created?

rusefi, 2020-06-17:
There are 7 unrelated bundles but only one date file to change. I would try to matrix bundles to get those faster.

chuckwagoncomputing, 2020-06-17:
So basically like artifacts are build now, only change the file first.

rusefi, 2020-06-17:
I am not sure, we might be misunderstanding.

At the moment bundles are build consequentially, say 7x10 minutes = 70 minutes

my preference for bundles (which is technically different ticket so sorry for the mix-up) would be to build all 7 bundles in 10 minutes by leveraging more free GH hardware

fair point that we want first build of the day already to contain proper date. technically if all 7 builds would attempt to change to same date I assume git would only make one change anyway so we are good.

maybe it would be simpler to separate those two requests first? we can later integrate once we develop separately to simplify things.

PR #1504 (2020-06-17): Workflow to write the date, once a day
This should fix #1415
closed 2020-06-17 by rusefi