rusefi/rusefi #1417: what is the root cause of rusEFI spi not accepting so many SD cards?

rusefi, 2020-05-10:
do we know if Arduino has a better success rate via same SPI? could it be that we have defects in rusEFI?

rusefi, 2020-05-10:
@stefanst FYI

mck1117, 2020-05-10:
Apparently some newer SD/SDHC cards have dropped support for SPI mode.

https://forum.4dsystems.com.au/node/1869

>  "Please note that Kingston Technology does not manufacture and subsequently test its range of flash card products to be used for the purpose of being addressed under the SPI mode."

rusefi, 2020-05-10:
Maybe ChipGenius is a tool to look at card ID? I have not found better options to replace ReadCardInfo.zip from above link

also Frankenso to Arduino comparison would still make sense :)

rusefi, 2020-05-10:
there is 
[image]

there is uSD-4GB-Industrial 
[image]

rusefi, 2020-05-15:
Many good for us cards shown at https://rusefi.com/forum/viewtopic.php?p=36528#p36528

closed 2020-05-15 by rusefi