rusefi/rusefi #1420: We are concerned about TS Lite config update handling

rusefi, 2020-05-11:
@mck1117 @OrchardPerformance please provide more details

rusefi, 2020-05-11:
EFI says "The difference dialog has not been intentionally removed. I tried to reproduce this in the light edition and I am getting prompted each time.It would seem there is some other factor here."

mck1117, 2020-05-11:
Problem:
- In paid version of TS, upon connecting to the ECU, TS will compare the configuration stored on the ECU with the config stored in TS.  The user can then pick which one they want to keep, either sending TS's config to the ECU, or replacing TS's config with that from the ECU.  
- At some point in recent history this stopped working in the free version of TS, but not the paid version

OrchardPerformance, 2020-05-12:
To add further detail, I have tested this on 3 different computers, win 10 and win 7, tried with an MRE and a Frankenso. 
TS versions tried are TunerStudio MS Lite 3.0.29 (working ok) but 3.1 onward is not ok.
I don't have an installer for any version between 3.0.29 and 3.1 to test but I think this began with the January release of 3.1.

Issue has presented as a simple overwrite of setting/tune on connection, no dialog appears.

rusefi, 2020-05-13:
https://www.youtube.com/watch?v=PrHSWcGMvcU

mck1117, 2021-03-19:
this is indeed fixed in newer versions of TS

closed 2021-03-19 by mck1117