rusefi/rusefi #1426: TLE8888 outputs switched on during MCU reset on MRE

dron0gus, 2020-05-12:
1. run self stimulation.
2. press Reset button on MRE
3. INJ channels enabled for short period of time right after releasing reset button

This issue partially fixed in 0.5.0 HW. INJ and IGN channels have dedicated EN signals that are driven by STM32.
For previous revision we need to set INJ gpios to safe state. Input with Pull-up that is used now for all pins is not safe in this case.
Seems we need to set proper "safe" state for each GPIO individually. For each board.

mck1117, 2020-05-13:
This doesn't save you in case of reset.  We have no control over pins in reset.  If you update firmware, it might pin the injectors on.

dron0gus, 2020-05-13:
During reset (while reset button is pressed) - all pins are in input mode with pull-up and pull-down disabled. And TLE8888 has small pull-down current on inputs (25..100uA) - we are safe.
Early init code set all pins to inputs with pull-up enabled. This cause high level on TLE8888 inputs.
This is not safe.
Code do a loooooot of initialization. That takes a looooot of time.
We are still not safe. Outputs enabled.
rusEFI code grabs pins and configures it it proper mode.
We are safe.
BUT. If due to some configuration mismatch one of TLE8888 pins is not used - it left in unsafe state.
During firmware update (I hope) all pins unused by bootloader are in input with no pull-up/down mode. This is safe.

mck1117, 2020-05-14:
Why do we not just "input pulldown" every pin in `board.h`?

dron0gus, 2020-05-14:
Input pull down is not safe for active-low signals. Like SPI CSs, ETB enable signal and so on.
I see no reason for common magic safe state for ALL pins.
Each pin should have its own safe state. That we should set in early init code.

mi-hol, 2023-04-01:
@dron0gus as you pointed out this is a **product safety** related issue, it should have gotten **high priority to fix**.

Are you in a position to comment on current status?