rusefi/rusefi #1429: BUG? handleCrc32Check does not use 'offset' parameter

rusefi, 2020-05-13:
offset parameter of handleCrc32Check method is used for logging only, is that a defect?

```void handleCrc32Check(ts_channel_s *tsChannel, ts_response_format_e mode, uint16_t pageId, uint16_t offset,
		uint16_t count) {
	UNUSED(pageId);

	tsState.crc32CheckCommandCounter++;

	count = getTunerStudioPageSize();

	scheduleMsg(&tsLogger, "CRC32 request: offset %d size %d", offset, count);

	uint32_t crc = SWAP_UINT32(crc32((void * ) getWorkingPageAddr(), count));

	scheduleMsg(&tsLogger, "CRC32 response: %x", crc);

	sr5SendResponse(tsChannel, mode, (const uint8_t *) &crc, 4);
}
```

rusefi, 2020-05-21:
Another Easter egg:

```
	;
	; TODO: interesting - 0x36B8 equals 14008 which looks like pageSize at some point in the past? is this a defect here - should
	; we have current pageSize or (pageSize + 8) here?
	;
   crc32CheckCommand   = "k\x00\x00\x00\x00\x36\xB8"
```

rusefi, 2020-05-21:
this defect is why TS always reads whole page on start-up, we can speed-up connect by fixing this! let me give this a try.

rusefi, 2020-05-21:
maybe false alarm. looking deeper...

rusefi, 2020-05-21:
good news: original issue description is not an issue just a mess - mess is now less messy

bad news: CRC on start-up is not happy for some reason?!

rusefi, 2020-06-19:
Still an issue :(

```
6/18/20 11:17:56 PM :Info: Communicating with sig:rusEFI v1.2020.4, rusEFI v20200619@23544
6/18/20 11:17:56 PM :Debug: Received Instruction: Read All Tune Data, Page: 0
6/18/20 11:17:56 PM :Debug: controllerOnline = 1.0
6/18/20 11:17:56 PM :Info: CRC Mismatch, will need to refresh page from controller 
6/18/20 11:17:56 PM :Info:    CRC from controller page 1:x4B xA9 xD8 x9C                                                       K...
6/18/20 11:17:56 PM :Info:    Local Data CRC for page 1: x21 x20 x36 x1D                                                       !.6.
6/18/20 11:17:56 PM :Info: Activating page: 1
6/18/20 11:17:59 PM :Debug: Read page time: 2328ms.
6/18/20 11:17:59 PM :Info: Checksum page 1 data read: x4B xA9 xD8 x9C                                                       K...
```

rusefi, 2020-06-19:
Well, I am out of ideas

We now have two command line tools to calculate CRC32
get_image_tune_crc to CRC binary tune file produced by console
get_xml_tune_crc to read XML

I've used get_xml_tune_crc to read TS own XML tune and CRC matches binary tune. What is the other CRC value in TS logs coming from I have no idea?!

rusefillc, 2020-08-24:
@mck1117 time to close?

mck1117, 2020-08-24:
yes, time to close, my PR the other day fixes this

closed 2020-08-24 by rusefillc