rusefi/rusefi #1454: [Feature Request] Transmission Control

Gouster4, 2020-05-19:
Hello. I found this on spreduino forums:
http://speeduino.com/forum/viewtopic.php?t=592
It explain how simple is interacting with automatic transmissions (at least some).

And i also found this solution for speeduino:
https://speeduino.com/forum/viewtopic.php?f=15&t=1827/

Can someone get it working with rusEfi?

Thanx.

rusefi, 2020-05-20:
@Gouster4 are you available to test any code if code would appear? what transmission do you have available for the testing?

Gouster4, 2020-05-20:
I have 1996 mustang v6 with
4R70W or AODE (not sure now).

I ordered esp32 dev board just dew days ago from ebay. And im planning to make own custom board design to fit rusefi into my ecu case... So i can test it, but i dont know when my rusefi will be prepared.

Gouster4, 2020-05-20:
Can whole tranamissin control be controlled by same  stm32 board?
On speeduino its separated because it uses arduino mega which is less powerfull than stm32, and has less pins.

rusefi, 2020-05-20:
absolutely can be controlled by same stm32, we have CPU and RAM available for this for sure

please post a thread on the forum, we are getting further away from original subject but see Trabant question on https://github.com/rusefi/rusefi/wiki/Hardware FAQ

Autohome2, 2020-05-20:
The project you linked to was designed to be used as a standalone or in
partnership with speeduino hence why it was on its own mcu board (a 2560
mega)

Gouster4, 2020-05-20:
> The project you linked to was designed to be used as a standalone or in partnership with speeduino hence why it was on its own mcu board (a 2560 mega)
> […](#)

This project i linked had its own arduinoega to control tranamission with its own firmware, and also had some additional code for speeduino's firmware to add support...

I was just thinking if u can integrate whole that transmission control to rusefi, to save cost for another board and also space which is important when trying to fit rusefi into original ecu case.

Autohome2, 2020-05-20:
> ACXTBWBAKOP2G7OIQZ4TOP3RSN6OPANCNFSM4NE47FBQ .

That's exactly what i said, i know this as it was my project :-)

Personally even if the engine ecu has the capability to run a transmission
i feel a independent ecu is a better option, like many OEM .

Gouster4, 2020-05-20:
> That's exactly what i said, i know this as it was my project :-) Personally even if the engine ecu has the capability to run a transmission i feel a independent ecu is a better option, like many OEM .

Makes sense...

Those 2 commits, they are implementation of support for rufusefi (that part of code which was intended to speeduino)?
And after connecting that standalone transmission control board with its original firmware, it should work now?

rusefi, 2020-05-20:
@Gouster4 I am not sure what your comments are referring to I am now confused.

Gouster4, 2020-05-20:
> @Gouster4 I am not sure what your comments are referring to I am now confused.

https://github.com/rusefi/rusefi/commit/b9fedcab5aed0a908a41dddef9b6fa2c6d9cdc4d
and 
https://github.com/rusefi/rusefi/commit/e8ccb90fb6208247e60ea8366400171cb0f370d1

Those commits are something like this:
https://github.com/dvjcodec/dxControl-Gears
For rusefi? 
To support this aditional firmware on separate controller? https://github.com/dvjcodec/SCG-ATC

Or did i understood it wrong?

rusefi, 2020-05-20:
these commits are the first 1% of the implementation. As long as this ticket is open there is no reason to assume that rusEFI has TCU working. It scares me that you assume that this is a 5 minute job.

Gouster4, 2020-05-20:
> these commits are the first 1% of the implementation. As long as this ticket is open there is no reason to assume that rusEFI has TCU working. It scares me that you assume that this is a 5 minute job.

Thanx for info... Tbh, i have no clue how much takes doing such automatic transmission control implementation.

mck1117, 2020-05-22:
> Thanx for info... Tbh, i have no clue how much takes doing such automatic transmission control implementation.

One major blocker is that none of the developers have an engine that runs rusefi connected to a computer controlled automatic.  They're all manual, mechanical/hydraulic automatics, or running on their own (self contained) computer.  I am happy to let someone pay me to buy a 4L60E/4L80E and driveshaft, swap it in to my car, develop the code, then put my T56 back in place once it's done, but I don't think you'd like my hourly rate for that.

OrchardPerformance, 2020-05-29:
Given the number of possible transmissions and individual things to deal with it feels like this is a separate module that receives its inputs from a rusEFI unit and not something that should really be in the ECU. 
I feel it would be best if we could make some of the outputs available via can for integration to external modules in future and a separate (but obviously linked) project should take on the hardware and software needs of the TCU.

rusefi, 2020-05-29:
I wonder if we can cover 80% of the requested transmissions with 20% of the effort?

At the moment I am kind of thinking to play with all this since I have two test mules. I would say we can make the decision based on what we learn from these two test mules.

The plan is to support both combined ECU+TCU mode and ECU<>CAN<>TCU right from the beginning since we kind of have everything needed for that with existing "SLAVE" mode implemented during the 2xMRE=M73 fun

OrchardPerformance, 2020-05-29:
My only concern is we load the ECU up with more tasks that are not really "engine" things and reduce the ability to do engine stuff. 
My thinking it is may be better to have a re ally good ECU and a good separate TCU as opposed to a mediocre all in one. 
Have we looked at what is needed to connect to some of the more common autobox controllers? 
If some of them (like the M73) simply need a CAN connection then it would be simple to just interface with the stock unit.

OrchardPerformance, 2020-05-29:
It does also open the question if an MRE with a TCU code on it would be able to work as the TCU unit? 
Would a TCU fork make sense? 
Or a set of TCU functions that are only compiled into an MRE TCU build be the ideal solution?

rusefi, 2020-05-29:
I am only interested in rusEFI TCU working in tight cooperation with rusEFI ECU. I have zero motivation to work on rusEFI TCU which is detached from rusEFI ECU.

rusefi, 2020-05-30:
[845re_introduction.pdf](https://github.com/rusefi/rusefi/files/4704971/845re_introduction.pdf)