rusefi/rusefi #1455: [Feature Request ] Add VE/AFR tables to Alpha-N strategy

125jdavis, 2020-05-19:
Alpha-N fueling algorithm doesn't have VE table or target AFR table. Currently the user tunes by changing target PW values in the TPS/RPM table. Useability would be much better if the same VE/AFR fueling  tables from speed density were carried over to Alpha-N.

rusefi, 2020-05-19:
@125jdavis how would this work?

MAP and MAF works by applying the math of target AFR and VE over measured air. How do we get measured air in Alpha-N?

125jdavis, 2020-05-19:
[image]

Similar to Speed Density, but VE and Pressure get lumped together into the TPS term.

mck1117, 2020-05-19:
Is this a mostly-duplicate of #1299?

125jdavis, 2020-05-21:
@mck1117 I think you would know best.

Updated equation for O2 mass = TPS * Baro Pressure * Displacement / (Gas Constant * Intake Temp)

mck1117, 2020-05-21:
Instead of using TPS directly, I think you want a "VE" value looked up from a table as a function of TPS and RPM.  So essentially the same as the speed density math, but with pressure removed, and VE looked up using TPS instead of MAP.

andreika-git, 2020-05-22:
> "VE" value looked up from a table as a function of TPS and RPM

And we already have it implemented:
[image]

rusefi, 2020-05-22:
@OrchardPerformance can you please document and add TS tooltip?

OrchardPerformance, 2020-05-24:
I don't think this function is quite what is being meant. 
This function changes the fuel tables Y axis to TPS when in speed density mode. 
But only when "TPS Based" is selected as an option in the "VE auto tune" settings. 

What Matt is referring to is making the standard AlphaN mode have a fuel table in VE rather than Ms pulse width. 
Either way it is all covered in the other ticket.

PR #1636 (2020-07-25): new alphan implementation
part 3 of #1299.

also fix #1455 

Benefits of this over the old time-based fuel map method:
- compensates for injector size
- uses AFR table
- computes usable airmass value (for ignition Y-axis if/when we get there)

Because it is a fundamentally different mode, it gets a new enum value.  The old one will be removed so that we can intentionally cause an error to signal to the user that they need to visit their VE table.
closed 2020-07-25 by rusefillc