rusefi/rusefi #1460: CJ125 mess

rusefi, 2020-05-20:
something is not right

Abricosvw, 2020-05-20:
```
2020-05-20_18_39_44_417: CommandQueue: SerialIO started
2020-05-20_18_39_44_950: PortHolder: Opening port: COM4
2020-05-20_18_39_47_321: EngineState: applyNonPersistentConfiguration()
2020-05-20_18_39_47_322: EngineState: initializeTriggerWaveform(TT_60_2_VW/20)
2020-05-20_18_39_47_323: EngineState: console RX on PC11
2020-05-20_18_39_47_323: EngineState: console TX on PC10
2020-05-20_18_39_47_324: EngineState: *** rusEFI LLC (c) 2012-2020. All rights reserved.
2020-05-20_18_39_47_325: EngineState: rusEFI v20200515@22918
2020-05-20_18_39_47_327: EngineState: *** Chibios Kernel:       5.1.0
2020-05-20_18_39_47_327: EngineState: *** Compiled:     May 19 2020 - 12:05:35
2020-05-20_18_39_47_328: EngineState: COMPILER=9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
2020-05-20_18_39_47_329: EngineState: UID=40004F 33355103 33303934
2020-05-20_18_39_47_330: EngineState: CORTEX_MAX_KERNEL_PRIORITY=2
2020-05-20_18_39_47_331: EngineState: STM32_ADCCLK=21000000
2020-05-20_18_39_47_331: EngineState: STM32_TIMCLK1=84000000
2020-05-20_18_39_47_332: EngineState: STM32_TIMCLK2=168000000
2020-05-20_18_39_47_333: EngineState: STM32_PCLK1=42000000
2020-05-20_18_39_47_333: EngineState: STM32_PCLK2=84000000
2020-05-20_18_39_47_334: EngineState: PORT_IDLE_THREAD_STACK_SIZE=32
2020-05-20_18_39_47_335: EngineState: CH_DBG_ENABLE_ASSERTS=1
2020-05-20_18_39_47_336: EngineState: CH_DBG_SYSTEM_STATE_CHECK=1
2020-05-20_18_39_47_337: EngineState: CH_DBG_ENABLE_STACK_CHECK=1
2020-05-20_18_39_47_341: EngineState: EFI_LOGIC_ANALYZER=1
2020-05-20_18_39_47_341: EngineState: EFI_TUNER_STUDIO=1
2020-05-20_18_39_47_342: EngineState: EFI_SIGNAL_EXECUTOR_SLEEP=0
2020-05-20_18_39_47_343: EngineState: EFI_SIGNAL_EXECUTOR_HW_TIMER=0
2020-05-20_18_39_47_344: EngineState: EFI_SHAFT_POSITION_INPUT=1
2020-05-20_18_39_47_395: EngineState: EFI_INTERNAL_ADC=1
2020-05-20_18_39_47_396: EngineState: led: CRITICAL status on PD14
2020-05-20_18_39_47_401: EngineState: applyNonPersistentConfiguration()
2020-05-20_18_39_47_401: EngineState: initializeTriggerWaveform(TT_60_2_VW/20)
2020-05-20_18_39_47_402: EngineState: MAP on PC0
2020-05-20_18_39_47_405: EngineState: MAF on PB0
2020-05-20_18_39_47_408: EngineState: TPS 1 Primary on PA0
2020-05-20_18_39_47_409: EngineState: Acc Pedal on PA7
2020-05-20_18_39_47_412: EngineState: VBatt on PA4
2020-05-20_18_39_47_413: EngineState: CLT on PA6
2020-05-20_18_39_47_413: EngineState: CJ125 UR on PC1
2020-05-20_18_39_47_417: EngineState: CJ125 UA on PC3
2020-05-20_18_39_47_417: EngineState: SPI clock on NONE
2020-05-20_18_39_47_418: EngineState: SPI master out on NONE
2020-05-20_18_39_47_419: EngineState: SPI master in  on NONE
2020-05-20_18_39_47_419: EngineState: SPI clock on PB13
2020-05-20_18_39_47_420: EngineState: SPI master out on PB15
2020-05-20_18_39_47_421: EngineState: SPI master in  on PB14
2020-05-20_18_39_47_421: EngineState: Fuel pump on ext:(null).-1
2020-05-20_18_39_47_426: EngineState: CAN TX on PB6
2020-05-20_18_39_47_426: EngineState: CAN RX on PB12
2020-05-20_18_39_47_465: EngineState: engine chart: Yes
2020-05-20_18_39_47_466: EngineState: engine chart size=300
2020-05-20_18_39_47_470: EngineState: Cam on PA2
2020-05-20_18_39_47_473: EngineState: Injector 1 on PE4
2020-05-20_18_39_47_478: EngineState: Injector 2 on PE2
2020-05-20_18_39_47_481: EngineState: Injector 3 on PE3
2020-05-20_18_39_47_484: EngineState: Injector 4 on PE1
2020-05-20_18_39_47_486: EngineState: Coil 1 on PD1
2020-05-20_18_39_47_488: EngineState: Coil 2 on PD6
2020-05-20_18_39_47_489: EngineState: Coil 3 on PD3
2020-05-20_18_39_47_491: EngineState: Coil 4 on PD4
2020-05-20_18_39_47_492: EngineState: EL accel th=5.00/mult=0.00
2020-05-20_18_39_47_495: EngineState: EL decel th=0.00/mult=0.00
2020-05-20_18_39_47_497: EngineState: TPS accel th=40.00/mult=0.00
2020-05-20_18_39_47_498: EngineState: beta=0.00/tau=0.00
2020-05-20_18_39_47_499: EngineState: EL accel th=5.00/mult=0.00
2020-05-20_18_39_47_501: EngineState: EL decel th=0.00/mult=0.00
2020-05-20_18_39_47_501: EngineState: TPS accel th=40.00/mult=0.00
2020-05-20_18_39_47_503: EngineState: beta=0.00/tau=0.00
2020-05-20_18_39_47_504: EngineState: FSIOpwm on PE0
2020-05-20_18_39_47_509: EngineState: FSIO_OUT_14 on PF13
[... 304 lines cut ...]
2020-05-20_18_41_19_447: EngineState: cj125: diag NOT GOOD
2020-05-20_18_41_19_448: EngineState: VM Short to GND
2020-05-20_18_41_19_448: EngineState: UN Short to GND
2020-05-20_18_41_19_448: EngineState: IA LOOKS GOOD
2020-05-20_18_41_19_448: EngineState: HR LOOKS GOOD
2020-05-20_18_41_19_449: EngineState: cj125 P=8.000000000 I=0.003000000 D=0.000000000
2020-05-20_18_41_19_449: EngineState: cj125 SPI_DEVICE_2 mosi=PB15
2020-05-20_18_41_19_449: EngineState: cj125 SPI_DEVICE_2 miso=PB14
2020-05-20_18_41_19_449: EngineState: cj125 SPI_DEVICE_2 sck=PB13
2020-05-20_18_41_19_450: EngineState: cj125: Check ident=0x60 diag=0xD0 init1=0x89 init2=0x10
2020-05-20_18_41_19_450: EngineState: cj125: diag NOT GOOD
2020-05-20_18_41_19_450: EngineState: VM Short to GND
2020-05-20_18_41_19_450: EngineState: UN Short to GND
2020-05-20_18_41_19_451: EngineState: IA Low Voltage
2020-05-20_18_41_19_451: EngineState: HR LOOKS GOOD
2020-05-20_18_41_19_451: EngineState: cj125: Loading stored calibration data (6940 2698)
2020-05-20_18_41_25_982: EngineState: WARNING: Check Tachometer Pulse per Rev!
2020-05-20_18_41_30_491: EngineState: confirmation_cj125_info:10
2020-05-20_18_41_30_491: EngineState: cj125: state=PREHEAT diag=0xFF (current Ua=2.109 Ur=0.415) (calibration Ua=1.694 Ur=0.658)
2020-05-20_18_41_30_492: EngineState: cj125: diag Looks great!
2020-05-20_18_41_30_493: EngineState: cj125 P=8.000000000 I=0.003000000 D=0.000000000
2020-05-20_18_41_30_511: EngineState: cj125 SPI_DEVICE_2 mosi=PB15
2020-05-20_18_41_30_513: EngineState: cj125 SPI_DEVICE_2 miso=PB14
2020-05-20_18_41_30_515: EngineState: cj125 SPI_DEVICE_2 sck=PB13
2020-05-20_18_41_35_929: EngineState: WARNING: Check Tachometer Pulse per Rev!

rusefi, 2020-05-20:
Above was with invalid settings - there was no UA divider in reality and analog divider was not set to ratio=2

Abricosvw, 2020-05-20:
```
2020-05-20_19_06_07_142: CommandQueue: SerialIO started
2020-05-20_19_06_07_659: PortHolder: Opening port: COM4
2020-05-20_19_06_09_558: EngineState: applyNonPersistentConfiguration()
2020-05-20_19_06_09_559: EngineState: initializeTriggerWaveform(TT_60_2_VW/20)
2020-05-20_19_06_09_560: EngineState: console RX on PC11
2020-05-20_19_06_09_561: EngineState: console TX on PC10
2020-05-20_19_06_09_562: EngineState: *** rusEFI LLC (c) 2012-2020. All rights reserved.
2020-05-20_19_06_09_563: EngineState: rusEFI v20200515@22918
2020-05-20_19_06_09_564: EngineState: *** Chibios Kernel:       5.1.0
2020-05-20_19_06_09_565: EngineState: *** Compiled:     May 19 2020 - 12:05:35
2020-05-20_19_06_09_566: EngineState: COMPILER=9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
2020-05-20_19_06_09_567: EngineState: UID=40004F 33355103 33303934
2020-05-20_19_06_09_567: EngineState: CORTEX_MAX_KERNEL_PRIORITY=2
2020-05-20_19_06_09_568: EngineState: STM32_ADCCLK=21000000
2020-05-20_19_06_09_569: EngineState: STM32_TIMCLK1=84000000
2020-05-20_19_06_09_570: EngineState: STM32_TIMCLK2=168000000
2020-05-20_19_06_09_577: EngineState: STM32_PCLK1=42000000
2020-05-20_19_06_09_578: EngineState: STM32_PCLK2=84000000
2020-05-20_19_06_09_584: EngineState: PORT_IDLE_THREAD_STACK_SIZE=32
2020-05-20_19_06_09_585: EngineState: CH_DBG_ENABLE_ASSERTS=1
2020-05-20_19_06_09_586: EngineState: CH_DBG_SYSTEM_STATE_CHECK=1
2020-05-20_19_06_09_586: EngineState: CH_DBG_ENABLE_STACK_CHECK=1
2020-05-20_19_06_09_587: EngineState: EFI_LOGIC_ANALYZER=1
2020-05-20_19_06_09_588: EngineState: EFI_TUNER_STUDIO=1
2020-05-20_19_06_09_588: EngineState: EFI_SIGNAL_EXECUTOR_SLEEP=0
2020-05-20_19_06_09_650: EngineState: EFI_SIGNAL_EXECUTOR_HW_TIMER=0
2020-05-20_19_06_09_651: EngineState: EFI_SHAFT_POSITION_INPUT=1
2020-05-20_19_06_09_652: EngineState: EFI_INTERNAL_ADC=1
2020-05-20_19_06_09_652: EngineState: led: CRITICAL status on PD14
2020-05-20_19_06_09_653: EngineState: applyNonPersistentConfiguration()
2020-05-20_19_06_09_654: EngineState: initializeTriggerWaveform(TT_60_2_VW/20)
2020-05-20_19_06_09_654: EngineState: MAP on PC0
2020-05-20_19_06_09_656: EngineState: MAF on PB0
2020-05-20_19_06_09_656: EngineState: TPS 1 Primary on PA0
2020-05-20_19_06_09_657: EngineState: Acc Pedal on PA7
2020-05-20_19_06_09_658: EngineState: VBatt on PA4
2020-05-20_19_06_09_658: EngineState: CLT on PA6
2020-05-20_19_06_09_659: EngineState: CJ125 UR on PC1
2020-05-20_19_06_09_660: EngineState: CJ125 UA on PC3
2020-05-20_19_06_09_660: EngineState: SPI clock on NONE
2020-05-20_19_06_09_661: EngineState: SPI master out on NONE
2020-05-20_19_06_09_662: EngineState: SPI master in  on NONE
2020-05-20_19_06_09_662: EngineState: SPI clock on PB13
2020-05-20_19_06_09_664: EngineState: SPI master out on PB15
2020-05-20_19_06_09_664: EngineState: SPI master in  on PB14
2020-05-20_19_06_09_665: EngineState: Fuel pump on ext:(null).-1
2020-05-20_19_06_09_666: EngineState: CAN TX on PB6
2020-05-20_19_06_09_666: EngineState: CAN RX on PB12
2020-05-20_19_06_09_691: EngineState: engine chart: Yes
2020-05-20_19_06_09_695: EngineState: engine chart size=300
2020-05-20_19_06_09_696: EngineState: Cam on PA2
2020-05-20_19_06_09_701: EngineState: Injector 1 on PE4
2020-05-20_19_06_09_702: EngineState: Injector 2 on PE2
2020-05-20_19_06_09_703: EngineState: Injector 3 on PE3
2020-05-20_19_06_09_703: EngineState: Injector 4 on PE1
2020-05-20_19_06_09_704: EngineState: Coil 1 on PD1
2020-05-20_19_06_09_711: EngineState: Coil 2 on PD6
2020-05-20_19_06_09_712: EngineState: Coil 3 on PD3
2020-05-20_19_06_09_713: EngineState: Coil 4 on PD4
2020-05-20_19_06_09_713: EngineState: EL accel th=5.00/mult=0.00
2020-05-20_19_06_09_717: EngineState: EL decel th=0.00/mult=0.00
2020-05-20_19_06_09_718: EngineState: TPS accel th=40.00/mult=0.00
2020-05-20_19_06_09_719: EngineState: beta=0.00/tau=0.00
2020-05-20_19_06_09_720: EngineState: EL accel th=5.00/mult=0.00
2020-05-20_19_06_09_720: EngineState: EL decel th=0.00/mult=0.00
2020-05-20_19_06_09_721: EngineState: TPS accel th=40.00/mult=0.00
2020-05-20_19_06_09_722: EngineState: beta=0.00/tau=0.00
2020-05-20_19_06_09_722: EngineState: FSIOpwm on PE0
2020-05-20_19_06_09_723: EngineState: FSIO_OUT_14 on PF13
[... 307 lines cut ...]
2020-05-20_19_09_31_688: EngineState: calc 2: action 2 value 0.00
2020-05-20_19_09_31_688: EngineState: calc 3: action RPM value 1200.00
2020-05-20_19_09_31_689: EngineState: calc 4: action 1 value 0.00
2020-05-20_19_09_31_710: EngineState: calc 5: action 3 value 1.00
2020-05-20_19_09_31_712: EngineState: calc 6: action 7 value 1.00
2020-05-20_19_09_31_714: EngineState: setPin pump on
2020-05-20_19_09_39_404: EngineState: confirmation_cj125_info:10
2020-05-20_19_09_39_404: EngineState: cj125: state=PREHEAT diag=0xFF (current Ua=0.400 Ur=0.391) (calibration Ua=0.371 Ur=0.654)
2020-05-20_19_09_39_404: EngineState: cj125: diag Looks great!
2020-05-20_19_09_39_405: EngineState: cj125 P=8.000000000 I=0.003000000 D=0.000000000
2020-05-20_19_09_39_405: EngineState: cj125 SPI_DEVICE_2 mosi=PB15
2020-05-20_19_09_39_405: EngineState: cj125 SPI_DEVICE_2 miso=PB14
2020-05-20_19_09_39_406: EngineState: cj125 SPI_DEVICE_2 sck=PB13
2020-05-20_19_09_41_407: EngineState: WARNING: Check Tachometer Pulse per Rev!
2020-05-20_19_09_50_648: EngineState: confirmation_cj125_info:10
2020-05-20_19_09_50_649: EngineState: cj125: state=PREHEAT diag=0xFF (current Ua=0.785 Ur=0.487) (calibration Ua=0.371 Ur=0.654)
[... cut ...]
2020-05-20_19_09_50_649: EngineState: cj125 P=8.000000000 I=0.003000000 D=0.000000000
2020-05-20_19_09_50_649: EngineState: cj125 SPI_DEVICE_2 mosi=PB15
2020-05-20_19_09_50_649: EngineState: cj125 SPI_DEVICE_2 miso=PB14
2020-05-20_19_09_50_650: EngineState: cj125 SPI_DEVICE_2 sck=PB13
2020-05-20_19_09_51_361: EngineState: WARNING: Check Tachometer Pulse per Rev!
2020-05-20_19_09_57_882: EngineState: cj125 ERROR: Sensor overheating.
2020-05-20_19_09_57_882: EngineState: cj125: Controller Shutdown!
2020-05-20_19_10_01_401: EngineState: WARNING: Check Tachometer Pulse per Rev!

mck1117, 2021-05-31:
do we give up on cj125?

rusefillc, 2021-05-31:
We are pretty close to giving up but let's just ignore it for now

closed 2024-01-04 by rusefillc