rusefi/rusefi #1466: "TLE8888 SR Unexpected response" does not work as intended

rusefi, 2020-05-22:
somehow this validation works for me but not for two users?

rusefi, 2020-05-22:
[image]
[image]

OrchardPerformance, 2020-05-23:
I think this is the same as I saw yesterday as well. 
I had a response of "0" from an older V0.3 MRE board 
https://files.slack.com/files-pri/T4NJHQ8QZ-F014EM60HH7/image.png

rusefi, 2020-05-23:
Looks like it actually works AS intended! Looks like it was a proper alert in case of Jacob - TLE was in fact not being powered!

closed 2020-05-23 by rusefi

OrchardPerformance, 2020-05-23:
I am sure it was not powered in my case, I thought that was the reason for the lack of response.

rusefi, 2020-05-23:
@OrchardPerformance below 7v vbatt it should not even try to communicate

mck1117, 2020-05-23:
Should the error message be something more useful?
```
if (voltage > 7) {
    bool success = attempt_comms();
    if (!success) {
         error("please check your wiring/tle8888/etc, something is wrong");
    }
}
```

commit 44b5e38d (2020-05-23): "TLE8888 SR Unexpected response" does not work as intended fix #1466