rusefi/rusefi #1467: move gcov CI to github actions

rusefi, 2020-05-22:
at the moment gcov requires linux, windows tool chain did not work for me so far

at the moment CI runs in a VM linux on a box which does too many things in parallel already. let's try moving gcov CI to GH actions

open question how artifact would be consumed and published on current location but first step let's produce the artifact

chuckwagoncomputing, 2020-06-16:
run_gcov.sh uses variables RUSEFI_DOXYGEN_FTP_USER, RUSEFI_DOXYGEN_FTP_PASS, and RUSEFI_FTP_SERVER.
You could put these values in Github secrets, and we could pass them to the run_gcov.sh script, which would be modified to accept them as arguments.

chuckwagoncomputing, 2020-06-16:
Like this:
https://github.com/chuckwagoncomputing/rusefi/blob/gcovaction/.github/workflows/build-unit-tests.yaml

ci_gcov.sh is just the modified version of run_gcov.sh.

rusefi, 2020-06-16:
https://help.github.com/en/actions/configuring-and-managing-workflows/creating-and-storing-encrypted-secrets sweet let me do this

rusefi, 2020-06-16:
three secrets added

rusefi, 2020-06-16:
next issue

cp: cannot stat '../build/obj/*': No such file or directory
./unit_tests/ci_gcov.sh: line 21: lcov: command not found
./unit_tests/ci_gcov.sh: line 22: genhtml: command not found
./unit_tests/ci_gcov.sh: line 24: ncftpput: command not found

chuckwagoncomputing, 2020-06-16:
yes, I'm still working on it

rusefi, 2020-06-16:
@chuckwagoncomputing 
current non-GH jenkins for GCOV is also not Windows it uses run_gcov.sh

new ci_gcov.sh reminds me of run_gcov.sh

what is the probability of ci_gcov.sh duplicating parts of run_gcov.sh?

chuckwagoncomputing, 2020-06-16:
Should be able to close this now.

closed 2020-06-16 by rusefi