rusefi/rusefi #1468: GH action to produce DFU

rusefi, 2020-05-23:
GH actions are cool and we are already using .bin files!

let's produce .dfu files as well and use those as well :)

rusefi, 2020-05-23:
@mck1117 @dron0gus can you help?

need to find a xxx2dfu tool for linux

chuckwagoncomputing, 2020-06-16:
hex2dfu compiles and runs on linux, I don't have a board to test it on though.
There is also https://github.com/OpenMusicKontrollers/dfuse_pack/
I haven't tried it.

rusefi, 2020-06-16:
https://github.com/encedo/hex2dfu this one?

please get an stm32f407discovery or MRE to test on :) also need to script it into github action

chuckwagoncomputing, 2020-06-16:
I used the rusefi/hex2dfu fork, as I understood it had a bug fix.

chuckwagoncomputing, 2020-06-16:
The workflows aren't showing up in my actions tab in my fork for some reason, so I can't test the workflow, so I guess I'll just go ahead and submit a PR.

chuckwagoncomputing, 2020-06-16:
Looks like it made a DFU, now someone with hardware ought to make sure it flashes correctly.

rusefi, 2020-06-16:
@OrchardPerformance can you please test?

rusefi, 2020-07-19:
I assume this works

closed 2020-07-19 by rusefi