rusefi/rusefi #1472: mass air flow output channel scaling needs help

mck1117, 2020-05-24:
[image]

We can't handle over 655.35 kg/hr of mass flow.  My engine flows this amount at 4200 rpm, and then 50% more at 6500.

OrchardPerformance, 2020-05-24:
Yep that's a problem. 
Just for the log or internal to the firmware? 
Do we need to start using g/s in the ECU and outputting the kg/h as a single decimal place?

rusefi, 2020-05-24:
not internal to ecu for sure

we can just change binary logging scale by any factor

mck1117, 2020-05-24:
right, should have clarified, this is just a problem with the output channel scaling.  The ECU uses float internally so it's not a problem.

mck1117, 2020-07-07:
fixed in #1566

closed 2020-07-07 by mck1117