rusefi/rusefi #1477: Sometimes legacy DFU does not work

rusefi, 2020-05-31:
Some users report legacy DFU does not work while cube DFU works for them. Weird.

At this point unclear what is this about.

I personally have both Windows 7 and Windows 10 working perfectly with legacy DFU driver and console.

See also #1170

SVeilleux9, 2020-06-01:
I have tested Ubuntu 20.04 with dfu-util version 0.9. I had no issues with the following command. 
`sudo dfu-util -a 0 -D rusefi_proteus-f4.dfu`

rusefi, 2020-06-01:
We had a session with @abricos and looks like we have an issue with switch to DFU in auto-port mode

with manually selected port software DFU mode switch works
with auto-detect port software DFU mode sometimes does not work

rusefi, 2020-06-02:
@abricos says stm32f427 MRE manual DFU from rusEFI console flashed fine
@abricos says stm32f427 MRE auto DFU from rusEFI console does NOT flash

rusefi, 2020-06-02:
so, two separate issues:
1) windows console auto-DFU with auto-port FIXED? 427 works just fine under Windows as long as in DFU mode

2) unix dfu util does not flash 427 while in DFU mode

rusefi, 2020-06-02:
@SVeilleux9 can you please paste _exact_  dfu-util output with version info?

@SVeilleux9 how _exactly_ did you install can you please spell out exact commands? from sources or from apt get?

SVeilleux9, 2020-06-02:
With verbose enabled. 
According to this page https://packages.ubuntu.com/search?keywords=dfu-util Ubuntu Focal (20.04) is on dfu-util 0.9-1.

```
sudo dfu-util -v -a 0 -D rusefi_proteus-f4.dfu 
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

DFU suffix version 11a
Match vendor ID from file: 0483
Match product ID from file: df11
dfu-util: Patching string descriptor 4 length (was 166, received 112)
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Memory segment at 0x08000000   4 x 16384 = 65536 (rew)
Memory segment at 0x08010000   1 x 65536 = 65536 (rew)
Memory segment at 0x08020000   7 x 131072 = 917504 (rew)
file contains 1 DFU images
parsing DFU image 1
Target name: EncedoKey
image for alternate setting 0, (1 elements, total size = 303168)
parsing element 1, address = 0x08000000, size = 303160
Erase   	[                         ]   0%            0 bytes   Poll timeout 100 ms
   Poll timeout 0 ms
Erase   	[=                        ]   4%        12288 bytes   Poll timeout 100 ms
   Poll timeout 0 ms
Erase   	[==                       ]   8%        24576 bytes   Poll timeout 100 ms
   Poll timeout 0 ms
Erase   	[===                      ]  12%        36864 bytes   Poll timeout 100 ms
   Poll timeout 0 ms
Erase   	[=====                    ]  20%        61440 bytes   Poll timeout 100 ms
   Poll timeout 0 ms
Erase   	[==========               ]  40%       122880 bytes   Poll timeout 100 ms
   Poll timeout 0 ms
Erase   	[=====================    ]  84%       256000 bytes   Poll timeout 100 ms
   Poll timeout 0 ms
Download	[                         ]   0%            0 bytes Download from image offset 00000000 to memory 08000000-080007ff, size 2048
   Poll timeout 104 ms
   Poll timeout 0 ms
.
.
.
 Download from image offset 0004a000 to memory 0804a000-0804a037, size 56
   Poll timeout 104 ms
   Poll timeout 0 ms
done parsing DfuSe file
```

rusefillc, 2023-01-16:
We now only use "newer" driver.

closed 2023-01-16 by rusefillc