rusefi/rusefi #1482: .logicdata files are a bit broken

rusefi, 2020-06-05:
we can now produce .logicdata files!

but Saleae tool crashes while attempting to open those :(

rusefi, 2020-06-05:
[image]

rusefi, 2020-06-05:
[logs.zip](https://github.com/rusefi/rusefi/files/4738114/logs.zip)

@andreika-git

rusefi, 2020-06-05:
[image]

rusefi, 2020-06-05:
I think I am using the latest version?

[image]

rusefi, 2020-06-05:
@andreika-git LogicdataStreamFileSandbox added and even that file is not well :(

[sandbox.zip](https://github.com/rusefi/rusefi/files/4738149/sandbox.zip)

rusefi, 2020-06-05:
Anyway https://support.saleae.com/faq/technical-faq/can-i-create-or-edit-logicdata-files :)

andreika-git, 2020-06-05:
The footer is not written, "@Override" problem: writeFooter(Writer) is not overridden by writeFooter()

rusefi, 2020-06-05:
Some progress was done - footer is now written either always or at least sometimes.

Sandbox file now works!

Longer real files still do not work :( can you please see if you can identify more issues? fresh logs attached

[logs.zip](https://github.com/rusefi/rusefi/files/4738814/logs.zip)

rusefi, 2020-06-06:
Now with event buffer still does not work :( 

[logs.zip](https://github.com/rusefi/rusefi/files/4740174/logs.zip)

rusefi, 2020-06-06:
@andreika-git 

[image]

andreika-git, 2020-06-06:
The timestamps should be 0-based (like from the start). Maybe just try to subtract the first timestamp from the remaining ones?

andreika-git, 2020-06-08:
#1484

commit ae712454 (2020-06-08): .logicdata files are a bit broken fix #1482
closed 2020-06-08 by rusefi