rusefi/rusefi #1488: Trigger decoding issues (60-2)

960, 2020-06-11:
The last months, I have not made any firmware-versions running without trigger-issues.
The car just stops and throws trigger-errors.

I have a older firmware running without this problem.
The car also runs perfect with both Link-Ecu and oem ecu.

It makes no difference at all if I download firmware, build it, ICU vs Exti etc..

The code it usually throws are these:

CUSTOM_OBD_TRG_DECODING = 6035,
CUSTOM_SYNC_ERROR = 6036,
CUSTOM_SYNC_COUNT_MISMATCH = 6037,

The trigger-logger in TS keeps showing negative values,but I dont know if thats related.

I attach a fresh driving log from both the older firmware that works + one newer that engine stops and throws these errors.
Both are done today with debug set as trigger-counter

[Logs.zip](https://github.com/rusefi/rusefi/files/4766496/Logs.zip)
 

[image]

mck1117, 2020-06-12:
what hardware/engine are these?

mck1117, 2020-06-12:
and how modified is the firmware? does it reproduce with 100% vanilla firmware from official bundle?

rusefi, 2020-06-12:
[image]

AFR, fuel logs - do these look healthy? @OrchardPerformance

960, 2020-06-12:
> what hardware/engine are these?

It's the A4 1.8t with the same old hardware I had all the time.

It makes no difference if I use the mre F7 bundle or try other build-settings/flags.

mck1117, 2020-06-12:
please share your configuration (TS debug package zip)

960, 2020-06-12:
I'll try to get it done later.. This is the tune 

[Audi_A4.zip](https://github.com/rusefi/rusefi/files/4769568/Audi_A4.zip)

960, 2020-06-12:
Also, it doesnt start again with just trying to start it in gear again, even if I drop rpm to 0.
I have to reduce speed enough to get the rpm down to around 1000 rpm for it to start

960, 2020-06-12:
2020-06-12_11_02_56_146: EngineState: rpm=711 time=379 index=0: gap=3.697 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_147: EngineState: error: synchronizationPoint @ index 46 expected 116/0/0 got 25/0/0
2020-06-12_11_02_56_148: EngineState: rpm=711 time=379 index=0: gap=0.029 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_149: EngineState: rpm=711 time=379 index=0: gap=0.711 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_150: EngineState: rpm=711 time=379 index=0: gap=1.476 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_150: EngineState: Emulating position sensor(s). RPM=2000
2020-06-12_11_02_56_151: EngineState: rpm=711 time=379 index=0: gap=8.662 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_152: EngineState: BURN in 4ms
2020-06-12_11_02_56_152: EngineState: rpm=711 time=379 index=0: gap=0.995 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_153: EngineState: rpm=711 time=379 index=0: gap=1.000 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_153: EngineState: CRC32 response: BBC91656
2020-06-12_11_02_56_154: EngineState: rpm=711 time=379 index=0: gap=0.999 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_154: EngineState: rpm=711 time=379 index=0: gap=1.004 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_155: EngineState: rpm=711 time=379 index=0: gap=0.996 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_156: EngineState: rpm=711 time=379 index=0: gap=0.999 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_156: EngineState: rpm=711 time=379 index=0: gap=1.003 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_157: EngineState: rpm=711 time=379 index=0: gap=0.997 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_158: EngineState: rpm=711 time=379 index=0: gap=1.002 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_159: EngineState: rpm=711 time=379 index=0: gap=0.997 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_159: EngineState: rpm=711 time=379 index=0: gap=0.999 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_160: EngineState: rpm=711 time=379 index=0: gap=0.999 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_161: EngineState: rpm=711 time=379 index=0: gap=1.003 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_162: EngineState: rpm=711 time=379 index=0: gap=1.000 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_162: EngineState: rpm=711 time=379 index=0: gap=0.995 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_163: EngineState: rpm=711 time=379 index=0: gap=1.000 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_164: EngineState: rpm=711 time=379 index=0: gap=1.000 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_165: EngineState: rpm=711 time=379 index=0: gap=1.002 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_166: EngineState: rpm=711 time=379 index=0: gap=0.996 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_167: EngineState: rpm=711 time=379 index=0: gap=0.999 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_168: EngineState: rpm=711 time=379 index=0: gap=1.000 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_168: EngineState: rpm=711 time=379 index=0: gap=1.002 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_169: EngineState: rpm=711 time=379 index=0: gap=0.997 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_169: EngineState: rpm=711 time=379 index=0: gap=1.003 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_170: EngineState: rpm=711 time=379 index=0: gap=0.995 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_170: EngineState: rpm=711 time=379 index=0: gap=0.999 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_171: EngineState: rpm=711 time=379 index=0: gap=1.004 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_171: EngineState: rpm=711 time=379 index=0: gap=0.996 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_172: EngineState: rpm=711 time=379 index=0: gap=1.002 expected from 1.600 to 4.000 error=Yes
2020-06-12_11_02_56_172: EngineState: rpm=711 time=379 index=0: gap=1.993 expected from 1.600 to 4.000 error=Yes

960, 2020-06-12:
It triggers the exact same codes in TS with simulator.

Firmware are the latest from bundle.

[image]

960, 2020-06-12:
It triggers the exact same codes in TS with simulator.

Firmware are the latest from MRE bundle.
Not sure what more I can provide that are relevant to the issue.

[image]

960, 2020-06-12:
Firmware from 15.04.20 runs without stopping.
Not sure exact date this issue started, but thats the version I have.

rusefi, 2020-06-12:
please try https://github.com/rusefi/rusefi/releases

https://github.com/rusefi/rusefi/releases/tag/20200507_cinco_de_mayo
https://github.com/rusefi/rusefi/releases/tag/20200513
https://github.com/rusefi/rusefi/releases/tag/20200522

960, 2020-06-12:
I am not able to upload tunes to any of these versions.
It's just errors, Data write failed, TS hanging for long time etc..
Whats that??

960, 2020-06-12:
Problem solved!! :-)
Just had a 2 hour drive without one single trigger-error.

Solution was to use files from 16.05.20, that was a partial update of the files from 23.04.20
(recall I didnt update them fully because some different bug)

mck1117, 2020-06-12:
so is there an actual bug somewhere? is there an official release that you see the issue on or not?

rusefi, 2020-06-12:
Well, reporting that 16.05.20 worked is not a solution.

960, 2020-06-12:
There is a bug somewhere for sure..
I dont know what more I can provide for you to find it.

It's been there for a while, and I have tried all kind if build-flags and even build from bundle several times.

I think the issue loading tune to mcu was the reason I did a partial update of those files

rusefi, 2020-06-12:
Well, at the moment I see this iteration as overall negative value

1) "partial update" makes no sense to me
2) very poor steps to reproduce the problem and overall problem description

Let's please value my time and make more effort. A possible solution would be to compile each day version and report results for each day. For some days into the past we probably have GH artifacts? Just as an example - see artifacts attached to https://github.com/rusefi/rusefi/runs/703977011

960, 2020-06-12:
This is the exact reason I stopped reporting issues..
It's just asking for lots of crap, even you know all the issues I have run into have turned out to be real.
The reply are a totally irelevant comment about unhealty AFR, and a request for a TS debug package..
Now I have two hours of logs with this pattern at debug I3

Lets please all value russians time, and waste our worthless time at compiling and reporting every day :-)

[image]

mck1117, 2020-06-12:
We can't help you until you can provide a report of how to reproduce this issue using:
1. official bundle
2. self stimulation

960, 2020-06-12:
Yes, the errors trigger with bundle and simulator. 
Try with around 3000 or something.

The errors uploading tune, worst after flashing started between 23.04 - 16.05 I think.
With those vanilla packs, I couldnt get it loaded no matter. TS freezes so bad I needed Task Manager to shut it..

It's not you helping me, but me helping you.
I have found a solution that works for me.

mck1117, 2020-06-12:
So current bundle trips it with self stimulator?

960, 2020-06-12:
Just had one running simulator at 3000 for a while.

At around 1000 Seconds it trips a row of 6036 and 6037.

[image]

960, 2020-06-13:
If you say this isnt a issue, we'll just close it.

I can keep the engine alive for a while at low steady rpm.
The thing that stops the engine right away are rpm acceleration.

I dont have any difference in firmware, and use the same config/outputchannels-struct.
Removing the additional files I have makes no difference.

I posted logs and tune.
I dont know what more explanation or additional information I can possible provide.

rusefi, 2020-06-13:
Stay tuned @mck1117 is investigating and at least some irregularities were observed

rusefi, 2020-06-14:
@mck1117 do we suspect that this is related to #1491?

960, 2020-06-14:
Matts change made it run without trigger errors in simulator.
Not tested in car. Maybe later today..

960, 2020-06-14:
Bad news.. Engine still stops.

[image]

rusefi, 2020-06-15:
@mck1117 says "I think the complete bug was:
Events A and B are scheduled very close together, T+10 and T+15
At T+10, we interrupt and execute A, which takes 4us.
At T+14, we check if more events are in the past that should have executed, and there are not.  This takes 2us.
At T+16, we reschedule the timer for the next event, which happens to be in the past.  In this case, we actually schedule the timer for 100us in the future (T+116, because of lateDelay), instead of when the event should've occurred, 1us in the past."

960, 2020-06-15:
I'll try it out with these changes.

I just remembered it has happened with the older firmware as well, but not often.

It always occure with fast RPM acceleration. 
For example in 1 or 2 gear, when boost kicks full.

rusefi, 2020-06-18:
"You should really have a look at TS config.
At later firmware versions, first upload of settings after flash are real broken.."

this is a sad example of sabotaging the process by throwing unrelated information on the ticket

this ticket should be ONLY about things related to Trigger decoding issues (60-2)

please create separate tickets for separate issues

rusefi, 2020-06-18:
#1491 is now merged into master, please try grab fresh logs with official build.

worst case we will wait for more tools to collect composite logs of your drive

960, 2020-06-18:
> "You should really have a look at TS config.
> At later firmware versions, first upload of settings after flash are real broken.."
>
> this is a sad example of sabotaging the process by throwing [...]

I would say it's pretty relevant to be able to upload settings to your builds for me to test it.
It writes a little, throws errors, I have to reconnect, write a little more over and over.

Your AFR comment are far less relevant. Even your post about this beeing relevant are less relevant.
We both know I have ran into tons of bugs, and all of them have turned out to be real.
Still same story over and over with you attacking, dont belive it, and just complaining.
You should learn to be thankful when someone trying to help you.
You have helped me with one thing:
I used to feel bad about bullying the geeks back in school-days..
Thanks to you, I dont have those feelings anymore.. I did the right thing.

On Topic:
The issue are not solved, the changes makes no differences.
I just discovered with my old working firmware, this issue starts if I enable spi2 and cj125.
Seems like something are balancing at the edge of overflowing something.

The shape of debug f6 are different at the running ones and those who dies.

Also:
None of the firmwares are able to start if you toggle ignition in speed and starting it in gear again.
Seems like they dont handle the large rpm-acceleration..
You have to reduce speed enough in highest gear to get it to about 1000 rpm for it to start.
You can try this yourself.

rusefi, 2020-06-18:
Fresh logs were requested, user has responded with a wall of text. Closing issue.

closed 2020-06-18 by rusefi

960, 2020-07-02:
I am not letting childish behavior keeping users from getting this fixed.
It's 100% for sure it's a firmware-bug.
It runs fine with ooem ecu, link ecu and same boards with speeduino.
I have even hooked up a max9926 to replace the tle8888, but it makes no difference.

You can see it clear at the below screenshot:

[image]

[2020-07-02_21.45.41.mlg.zip](https://github.com/rusefi/rusefi/files/4866784/2020-07-02_21.45.41.mlg.zip)

mck1117, 2020-07-02:
do you have a photo of the missing tooth on your flywheel?

960, 2020-07-02:
It's at the crank inside engine.
Looks like this:

[image]

rusefi, 2020-07-02:
It looks like we do not have enough logger speed to see gap for each tooth individually? Somehow we need better logs.

Maybe log fewer fields?

Maybe try "composite logs" in addition to normal logs? maybe logic analyzer? maybe wait until rusEFI console has better logging option. Maybe SD card logging with few fields? No idea how quick is SD card logging.

[image]

960, 2020-07-03:
I'll test a little and see if I can get better resolution.

Would it be a idea with filters that limits the allowed tooth time compared to the previous?
Like a enum with level 1, 2 and 3  (25%, 50% and 75%) that can be set in ts

960, 2020-07-03:
This has the best resolution I am able to get

[2020-07-03_16.17.14.mlg.zip](https://github.com/rusefi/rusefi/files/4870190/2020-07-03_16.17.14.mlg.zip)

rusefi, 2020-07-03:
I believe it's still not logging each tooth individually. I believe that at the moment we do not have a tool set to investigate this effectively.

[image]

960, 2020-07-03:
No, I see that.

But its clearly because the event-count just keep on above the expected ?

960, 2020-07-03:
This 0.3 width at count 1

[image]

960, 2020-07-03:
It keeps repeating in all logs instead of the tooth that should be 3

This value shows in tooth logger as well

[image]

rusefi, 2020-07-03:
Related to #1563

960, 2020-07-03:
This log shows it happening at idle even before the next sync?

[image]

[2020-07-03_16.17.14.mlg.zip](https://github.com/rusefi/rusefi/files/4870397/2020-07-03_16.17.14.mlg.zip)

960, 2020-07-04:
What setting do I use to make it sync at cam instead?
I'll try that and see if the issue are still there

It's 60-2 and the quick-start wheel, but I can cut off some teeth if thats easier.

rusefi, 2020-07-04:
quick-start wheel still not supported
if cutting i would be cutting 60/2 down to 30/2

before any cutting do you have the any logic analyzer like the $10 eBay? can you hook it up to processed VR signal between TLE8888 and stm32?

960, 2020-07-04:
Something like this :-)

[image]

I'll go out and give it a try.
I got new boards on the way, so these are getting retired.
New ones are without the TLE8888, and got footprint for both SPC5 and stm. They should be working with o5e, re and speeduino :-)

960, 2020-07-04:
Ok, here are a pulseview file from where it is stopping

[Stopping.zip](https://github.com/rusefi/rusefi/files/4873050/Stopping.zip)

960, 2020-07-04:
The toothwidth of 0.3 are toothduration after missing tooth / toothduration for missing teeth..

[image]

960, 2020-07-04:
Solution: Use chinese logic analyzer to read trigger signal? :-)

Each time this happen  index or eventcount keeps counting past the maximum 

[image]

rusefi, 2020-07-05:
(i am a bit behind, I absolutely plan to look into this properly - that you for the data)

960, 2020-07-05:
I think the issue are in this function

for (int i = 0;i<GAP_TRACKING_LENGTH;i++) {
				bool isGapCondition = cisnan(triggerShape->syncronizationRatioFrom[i]) || 
	(toothDurations[i] > toothDurations[i + 1] * triggerShape->syncronizationRatioFrom[i]
	&& toothDurations[i] < toothDurations[i + 1] * triggerShape->syncronizationRatioTo[i]);

				isSync &= isGapCondition;
The ratios are from 1.6 to 4 in Vw decoder.

Some places it pass 4. 

[image]

mck1117, 2020-07-05:
It's only looking at rising edges (in your case), so this is what it's measuring:
[image]

B is expected to be 3 * A with an N-2 wheel.

mck1117, 2020-07-05:
It looks like it's about 3.1-3.2 * A in your snip, which is totally fine.

960, 2020-07-05:
How hard would it be to add some cam signals to the decoder?

(In lack of other things to try :-) )

mck1117, 2020-07-05:
can you take a logic analyzer trace with both the crank signal AND the trigger error signal?  I want to see the exact instant the firmware decided something was wrong.

960, 2020-07-05:
I can place some lines in code to turn a pin(or more) high or low while driving with analyzer connected.

Where in code do you want them?

Isnt that done just something like this?

```
if (xxxxxx) {
				enginePins.xxxxxx.setValue(1);
			}

```

mck1117, 2020-07-06:
[image]

[39 more replies]