rusefi/rusefi #1492: code generator - reduce the gap between Windows and Linux

rusefi, 2020-06-14:
at the moment we have to manually sync gen_config.bat and gen_config.sh since all the parameters are hard-coded inside those platform-specific scripts

@ksmola got hit by that when .sh got outdated around firing order parameter

a proper fix would be to extract parameters into input file(s) so that one file could be modified and both scripts for both platforms would still work fine

chuckwagoncomputing, 2020-06-16:
If mingw is required anyway, couldn't we use just the .sh script?

chuckwagoncomputing, 2020-06-16:
I know this is low priority, but for what it's worth if you want to stick with having both .bat and .sh scripts, I can do the .sh side of things but I have know batch file knowledge whatsoever.

rusefi, 2020-06-16:
Using the same .sh on both Windows and Linux us actually an interesting idea!

commit 72dc218e (2020-06-16): Fix #1492 redux (#1496)

* use gen_config.conf to store flags, and gen_config.bat to call gen_config.sh

* Try sh.exe first, and fixed indentation
closed 2020-06-16 by rusefi