rusefi/rusefi #1499: build board bundles using GH Actions

rusefi, 2020-06-16:
At the moment slow Windows build server takes more than an hour to build all bundles for https://rusefi.com/build_server/ once case see difference in timestamps

between our new sh.exe cross-platform trick and GH secrets we shall migrate https://github.com/rusefi/rusefi/tree/master/misc/jenkins/compile_other_versions to bash and move to GH

a milestone would be to convert https://github.com/rusefi/rusefi/blob/master/misc/jenkins/build_working_folder.bat first

chuckwagoncomputing, 2020-06-16:
I think we should set up a secondary location to upload bundles to until this is completed. Maybe just a subdirectory of the build_server. Once everything is working properly, we can shut down the windows server and change the upload directory of the github action.

rusefi, 2020-06-16:
works for me - just subfolder in the same folder?

to test locally you would probably need a unit box and your own FTP server

chuckwagoncomputing, 2020-06-16:
The FTP server I have. Do you mean ECU by "unit box"?

rusefi, 2020-06-16:
@chuckwagoncomputing no sorry it was Unix box to run the scripts, but technically Cygwin would also work.

chuckwagoncomputing, 2020-06-16:
Ok gotcha. Most of my computers are Unix boxes. ;)

chuckwagoncomputing, 2020-06-17:
Should this be run with every push, or maybe once daily? Because we can do that.

rusefi, 2020-06-17:
I'd say every push, why not?

At the moment it slowly runs at every push. Usually when we push we want at least one of these bundles.

rusefi, 2020-06-19:
@chuckwagoncomputing see latest commit for some "build\rusefi.bin" WIN vs Linux drama

chuckwagoncomputing, 2020-06-19:
Oh good grief. :facepalm:

chuckwagoncomputing, 2020-07-08:
Should be fixed by #1583

closed 2020-07-08 by rusefi