rusefi/rusefi #1502: GH actions to invoke gen_config

rusefi, 2020-06-16:
At the moment we have Windows build server invoking gen_config_and_push.bat from https://github.com/rusefi/rusefi/tree/master/misc/jenkins which produces commits like https://github.com/rusefi/rusefi/commit/4460785120a4e2e3d59c7542ca1e93d34d5f134b

Where are two problems with that
1) Windows build box is under permanent stress especially after a push into GH
2) this requires manual intervention since github authentication expires or SOMETHING - current solution is not 100% reliable

Can we make this better by adding a GH action to do same?

chuckwagoncomputing, 2020-06-17:
This should be done every build I assume?

rusefi, 2020-06-17:
Every git merge/push I would say.

rusefi, 2020-06-18:
relevant key facts

1) GH is not 100% reliable and probably no SLA considering zero costs - there should a way to restore private Jenkins in case of emergency

2) we have to push into git since we want local build to benefit from automated generation

I think that means that some code reuse between GHA and Jenkins is needed. I think that means that first steps would be to convert more of existing Jenkins scripts into .sh for since at the moment main Jenkins is Windows (secondary Unix Jenkins in a VM which was used for gcov is deactivated now)

PR #1506 (2020-06-18): Run gen scripts and unify firmware, simulator, and unit_tests workflows
This should fix #1502.
Putting firmware, simulator, and unit test builds all in one workflow is necessary to ensure everything happens in the correct order.
closed 2020-06-18 by rusefi