rusefi/rusefi #1519: Only two .ini files got auto-generated

rusefi, 2020-06-19:
Expected behavior is for many .ini files to be refreshed together

[image]

rusefi, 2020-06-19:
One bug fixed

next bug to fix: need to revive kinetis generation!

```
#cd config\boards\kinetis\config
#!gen_config.bat
```
it apparently got lost while migrating from bat to .sh

FYI @chuckwagoncomputing

chuckwagoncomputing, 2020-06-19:
Why does kinetis have its own gen_config script?

rusefi, 2020-06-19:
@chuckwagoncomputing historical reasons mostly. it has some different parameters but overall could be part of the unified script.

One use-case is when kinetis was more actively developer it was easier to gen only kinetis

rusefi, 2020-06-19:
just fixed Proteus - that was a typo

still no idea why rusefi.ini not generated/commited @chuckwagoncomputing

rusefi, 2020-06-19:
rusefi.ini is the remaining issue @chuckwagoncomputing

rusefi, 2020-06-19:
makes no sense but it works! thank you!!!

PR #1520 (2020-06-19): Fix boards
This should fix #1519
closed 2020-06-19 by rusefi