rusefi/rusefi #1526: java: support compilation by both 8 and 11

rusefi, 2020-06-20:
My personal preference is to compile using java 8
the only change needed to compile by JDK 11 is 
```
            <!-- uncomment if you want to compile under java 11
            please remove the space between '-' and '-add-exports' it should be double-dash
                        <compilerarg line="- -add-exports java.xml/com.sun.org.apache.xerces.internal.parsers=ALL-UNNAMED"/>
                        <compilerarg line="- -add-exports java.xml/com.sun.org.apache.xml.internal.serialize=ALL-UNNAMED"/>
                        -->
```

technically it's possible to adjust build.xml to detect what JDK version is used and add those compilerarg conditionally

rusefillc, 2021-08-31:
@dron0gus says https://stackoverflow.com/questions/52502189/java-11-package-javax-xml-bind-does-not-exist 

could be as easy as adding jaxb into classpath but depends on gradle migration which was a bit postponed

rusefillc, 2021-11-05:
a proper healthy solution here would be to finish gradle migration (a lot was done already) and then make gradle conditional

rusefillc, 2022-09-01:
https://newrelic.com/resources/report/2022-state-of-java-ecosystem

says 50/50

[image]

commit e7a4db3e (2023-07-03): java: support compilation by both 8 and 11 fix #1526
commit e65e0f7e (2023-07-03): java: support compilation by both 8 and 11 fix #1526
closed 2023-07-03 by rusefillc