rusefi/rusefi #1528: UART DMA for "primary" connector

rusefi, 2020-06-22:
Need to refactor code to allow UART DMA + USB combination which is simply impossible right now

rusefi, 2020-06-22:
```
Index: firmware/console/connector_uart_dma.cpp
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
--- firmware/console/connector_uart_dma.cpp	(revision dfe5bcd6581004abaf9ad7e1100ddd3eb6d1903f)
+++ firmware/console/connector_uart_dma.cpp	(date 1592787995732)
@@ -61,7 +61,7 @@
 	iqObjectInit(&tsUartDma.fifoRxQueue, tsUartDma.buffer, sizeof(tsUartDma.buffer), NULL, NULL);
 
 	// start DMA driver
-	tsDmaUartConfig.speed = CONFIG(tunerStudioSerialSpeed);
+	tsDmaUartConfig.speed = 38400;//CONFIG(tunerStudioSerialSpeed);
 	uartStart(uartp, &tsDmaUartConfig);
 
 	// start continuous DMA transfer using our circular buffer
Index: firmware/config/stm32f4ems/efifeatures.h
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
--- firmware/config/stm32f4ems/efifeatures.h	(revision dfe5bcd6581004abaf9ad7e1100ddd3eb6d1903f)
+++ firmware/config/stm32f4ems/efifeatures.h	(date 1592786951790)
@@ -155,7 +155,8 @@
 #define EFI_MCP_3208 FALSE
 
 #ifndef EFI_HIP_9011
-#define EFI_HIP_9011 TRUE
+// disabling for now - DMA conflict with SPI1
+#define EFI_HIP_9011 FALSE
 #endif
 
 #ifndef EFI_CJ125
@@ -331,9 +332,11 @@
 
 // todo: start using consoleUartDevice? Not sure
 #ifndef EFI_CONSOLE_SERIAL_DEVICE
-#define EFI_CONSOLE_SERIAL_DEVICE (&SD3)
+//#define EFI_CONSOLE_SERIAL_DEVICE (&SD3)
 #endif
 
+#define CONSOLE_UART_DEVICE (&UARTD3)
+
 /**
  * Use 'HAL_USE_UART' DMA-mode driver instead of 'HAL_USE_SERIAL'
  *
@@ -343,10 +346,10 @@
  * in mcuconf.h
  */
 #define TS_UART_DMA_MODE FALSE
-#define PRIMARY_UART_DMA_MODE FALSE
+#define PRIMARY_UART_DMA_MODE TRUE
 
 //#define TS_UART_DEVICE (&UARTD3)
-#define TS_SERIAL_DEVICE (&SD3)
+//#define TS_SERIAL_DEVICE (&SD3)
 
 #define AUX_SERIAL_DEVICE (&SD6)
 
Index: firmware/config/stm32f4ems/mcuconf.h
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
--- firmware/config/stm32f4ems/mcuconf.h	(revision dfe5bcd6581004abaf9ad7e1100ddd3eb6d1903f)
+++ firmware/config/stm32f4ems/mcuconf.h	(date 1592786951792)
@@ -300,7 +300,7 @@
  */
 #define STM32_SERIAL_USE_USART1             TRUE
 #define STM32_SERIAL_USE_USART2             TRUE
-#define STM32_SERIAL_USE_USART3             TRUE
+#define STM32_SERIAL_USE_USART3             FALSE
 #define STM32_SERIAL_USE_UART4              FALSE
 #define STM32_SERIAL_USE_UART5              FALSE
 #define STM32_SERIAL_USE_USART6             TRUE

```

mck1117, 2022-06-23:
this is long fixed

closed 2022-06-23 by mck1117