rusefi/rusefi #1529: update_version.bat does not work on Windows

rusefi, 2020-06-22:
[image]

rusefi, 2020-06-22:
@chuckwagoncomputing

chuckwagoncomputing, 2020-06-22:
It works for me. Is git in your path? I'm using cygwin git.

rusefi, 2020-06-22:
I've changed PATH from standalone git to cygwin git, still does not work

[image]

chuckwagoncomputing, 2020-06-22:
Welp, the reason it was working for me was because I hadn't pulled this change in on my windows machine yet. PR incoming.

rusefi, 2020-06-22:
@chuckwagoncomputing it is OK to make a joke about testing less?

chuckwagoncomputing, 2020-06-22:
Is there a sheepish face emoji?

rusefi, 2020-06-22:
GH does not allow custom emoji but Slack does :)

closed 2020-06-22 by rusefi