rusefi/rusefi #1533: MRE no VSS I/O available for MK2+ Miata.

ByteVenom, 2020-06-23:
The NBx Miata ECUs receive a PWM wave from the instrument cluster for vehicle speed sensing. This is extremely useful for logging power runs, and more. Currently the drop down in the tuning menu (for input pin) does not have any options other than vital engine sensing pins. Is there any available IO that can handle this typ of input?

rusefi, 2020-06-24:
@stefanst  :)

rusefi, 2020-06-24:
@mck1117 says "I think any analog input can be used to read VSS signal. I would not use one of the temperature inputs, since they use a pullup, but any other analog in should work. I haven't really looked at the NB2 board at all yet, so I can't tell you where to go from here."

mck1117, 2020-06-24:
I didn't say that, @stefanst said that :)

closed 2026-01-28 by rusefillc