rusefi/rusefi #1534: NC IAT is not detected

johnceci-m3, 2020-06-23:
we have 4.95v on IAT which is clearly a wiring issue yet no IAT RED indicator

[image]

[image]

rusefi, 2020-06-23:
@mck1117 ?

mck1117, 2020-06-23:
and I assume you've rebooted the ECU since changing those settings?

rusefi, 2020-06-24:
@mck1117 yes 100% I was looking at this via teamviewer

mck1117, 2020-07-25:
did this ever get resolved?

rusefillc, 2020-07-25:
@mck1117 not to the best of my knowledge

mck1117, 2020-07-28:
@johnceci-m3 does this still reproduce for you? My "IAT failed" indicator comes on when I unplug it, so I'm not able to get this to happen.

mck1117, 2021-03-14:
No report of this happening again, closing.

closed 2021-03-14 by mck1117