rusefi/rusefi #1540: TS ignition settings for wasted spark not in sync with FW

stefanst, 2020-06-27:
In TS, when selecting 4 cylinder wasted spark under ignition settings, you can only select pins for "ignition pin 1" and "ignition pin 3" while the FW seems to be needing pins for "ignition pin 1" and "ignition pin 2". This leads to cars, set up apparently properly, to be running only on 2 cylinders, possibly leading to the destruction of catalytic converters.

rusefi, 2020-06-27:
What firing order is selected? "4 cylinder wasted" is not enough info to fully describe a setup for this.

Also this sounds too bad to be universally true? Either a recent bug or some details is affecting your experience @stefanst

rusefi, 2020-06-27:
Here is default MRE NB2 MAF tune which I believe starts and drivers the car

[image]

stefanst, 2020-06-27:
This approach (listing pin1 and pin3) used to work about a year ago
with old FW.
In my experience, this starts and runs the car only on 2 cylinders with
current FW. May be FW dependent. IIRC this did NOT work on FW 20200514
@Emil Adolphs <schmitzemil@live.com> has the same problem, but his FW
version is unknown as of now.. i have not tested with newer FW, since there
was no report of the issue having been adressed.

rusefi, 2020-06-27:
[image]