rusefi/rusefi #1542: "RPM Upper setting to deactivate IAC PID" not working as expected

stefanst, 2020-06-28:
The "RPM Upper setting to deactivate IAC PID" seems to effectively turn off IAC output for any value other than 0. Tried values from 1to 9000. Negative numbers are not allowed. The PID is still calculated as before, as can be seen from the debug fields, but the actual IAC value stays completely unmoved.

andreika-git, 2020-06-28:
Try setting 'useIacTableForCoasting' and 'iacCoasting' table, which are currently required for idlePidRpmUpperLimit to work correctly...

rusefi, 2020-06-28:
@stefanst I've make a change to TS project more info added into tooltips

@andreika-git do we need TS level field enabling logic like idlePidRpmUpperLimit only visible if useIacTableForCoasting == rtue?

Do we need firmware changes to handle invalid set of setting better with a warning or something?

@OrchardPerformance @mck1117 FYI

mck1117, 2020-06-28:
> Do we need firmware changes to handle invalid set of setting better with a warning or something?

We should trip critical error on any incompatible combination of options. Anything else is hard to diagnose.

stefanst, 2020-06-28:
I think TS settings should take care of these type issues. Much better making it impossible to pick than telling you "What you did there is not going to work" after the fact.
OTOH: in this particular case, it would appear that the setting for max rpm would make sense, even if we don't use a table for coasting. The table for coasting appears to be a solution to a specific problem for a small number of cases, whereas the max_rpm setting seems to have broader appliations.

mck1117, 2020-06-30:
yes, it's even better to selectively grey out/hide settings that don't make sense.  However, if we do that, the relevant settings MUST be physically near the on/off that greys them out.  Any other setup ruins the discoverability of the feature.

mck1117, 2021-11-29:
idle has long since been cleaned up

closed 2021-11-29 by mck1117