rusefi/rusefi #1551: Ignition and Injection pins not firing as intended

stefanst, 2020-07-01:
- Just set up a new MRE board with Miata NA main board.
- Uploaded 20200624 firmware.
- Spark settings as has been discussed here: Wasted spark, I can only set Ignition pin 1 and Ignition pin 3 (pin 2 is grayed out)
- Pin 1 is "ignition 1" and pin 3 is "ignition 3"
- And ONLY ignition THREE is flashing
- Also: Only the pins for one injector channel- channel 2 (we're paralleling two injectors for each channels, because we have batch injection) are firing.
[image]
[image]
[image]

Setting ignition to "Two wire wasted" seems to fix the issue. 
Once fixed, going back to "one wire batch" seems to still work.(needs confirmation).

stefanst, 2020-07-01:
This is the tune causing the issue.
[CurrentTune.zip](https://github.com/rusefi/rusefi/files/4855288/CurrentTune.zip)

rusefi, 2020-07-01:
expected trigger shape
https://rusefi.com/wiki/index.php?title=Manual:Software:Trigger#Mazda_Miata_NA

rusefi, 2020-07-01:
Engine sniffer shows only one of trigger channels - PA5. No data on PC6 trigger channel

tactical issue: HW issue, VR not converted to Hall
strategic issue: poor error handling in case of very broken trigger signal

mck1117, 2020-08-21:
I think this is fixed?