rusefi/rusefi #1553: idle_min through FSIO not working as intended

stefanst, 2020-07-01:
idle_min expression in FSIO formulas is as follows:
`ac_on_switch cfg_idleRpmPid_minValue cfg_idleRpmPid2_minValue if`

in console the following expression:
```
rpn_eval "ac_on_switch cfg_idleRpmPid_minValue cfg_idleRpmPid2_minValue if"
2020-07-01_10_57_32_970: EngineState: confirmation_rpn_eval "ac_on_switch cfg_idlerpmpid_minvalue cfg_idlerpmpid2_minvalue if":75
2020-07-01_10_57_32_975: EngineState: Parsing [ac_on_switch cfg_idlerpmpid_minvalue cfg_idlerpmpid2_minvalue if]
2020-07-01_10_57_32_975: EngineState: Evaluate result: 40.00
```
the evaluate result changes as expected with toggling the ac-in switch.

However, every once in a while, without any outside change, the messages show this:
```
2020-07-01_10_58_09_019: EngineState: state 34
2020-07-01_10_58_09_020: EngineState: idle settings: offset=30.000000000 P=0.00999 I=0.00999 D=0.00999 period=100ms
2020-07-01_10_58_09_020: EngineState: idle status: value=39.83 input=975.00/target=952.00 iTerm=10.16499 dTerm=-0.09999
2020-07-01_10_58_09_023: EngineState: WARNING: invalid expression for idle minValue
2020-07-01_10_58_09_113: EngineState: state 36
```
It would appear that state 36 is some sort of error state, because usually state is reported as 34.

Worse than that though is the actual output. The min value seems to have no effect on actual behavior of the IAC valve, when set via FSIO formulas. See attached log.  (IAC drop to 5). The minimum was set at 28.

[IAC drop to 5.zip](https://github.com/rusefi/rusefi/files/4858892/IAC.drop.to.5.zip)

rusefi, 2020-07-01:
@stefanst please attach tune
"state xxx" is idle state machine I will now improve this logging

stefanst, 2020-07-01:
[CurrentTune.zip](https://github.com/rusefi/rusefi/files/4861038/CurrentTune.zip)

rusefi, 2020-07-01:
tune shows a bit different expression but should be equally valid

[image]

stefanst, 2020-07-01:
Correct- "ac_on_switch not cfg_idleRpmPid_offset cfg_idleRpmPid2_offset if" (as in the posted tune) and "ac_on_switch cfg_idleRpmPid2_offset cfg_idleRpmPid_offset if" (as mentioned above) produce the same result using "rpn_eval". unfortunately, they also produce the same result running- no influence on actual minimum value.
Minimum value works as expected when not using FSIO.

rusefi, 2020-07-02:
@stefanst there are at list two defects there, working...

rusefi, 2020-07-02:
fixed applied @stefanst please give it a try

rusefi, 2020-07-06:
looks like that's related to idlePidRpmDeadZone

stefanst, 2020-07-07:
Fixed and tested

closed 2020-07-07 by stefanst