rusefi/rusefi #1563: rusEFI console produces invalid .logicdata files

rusefi, 2020-07-03:
I've attempted to record 60/2 trigger at 6000 RPM and this seems to produce a file which crashes Logic Analyzer on open :(

one of the attached .logicdata opens while the other does not

[logs.zip](https://github.com/rusefi/rusefi/files/4870298/logs.zip)

rusefi, 2020-07-03:
@andreika-git @mck1117 any interest?

rusefi, 2020-07-15:
@andreika-git ping any ideas?

rusefillc, 2020-07-20:
maybe related to #1621

rusefillc, 2021-04-04:
@andreika-git still an issue :(

[logs.zip](https://github.com/rusefi/rusefi/files/6255189/logs.zip)

rusefillc, 2021-04-05:
[logs_hellen.zip](https://github.com/rusefi/rusefi/files/6256186/logs_hellen.zip)

[logs_proteus.zip](https://github.com/rusefi/rusefi/files/6256187/logs_proteus.zip)

andreika-git, 2021-04-13:
Status update:
1) CSV writer doesn't write coil & injector flags (channels 5 and 6), so we cannot compare .csv to .logicdata adequately.
  To fix this, we need to add 2 more fields to TSHighSpeedLog.java (isCoil() and isInjector())
  Needless to say, we'll need a fresh set of .csv + .logicdata files for testing after that.
2) But, if we write "false" to channels 5+6, then .logicdata seems to work. You can test it with CompositeParser.java:
`boolean coil = false; /*EnumIniField.getBit(flags, 4);*/`
`boolean injector  = false; /*EnumIniField.getBit(flags, 5);*/`
  This is a temporary solution, until we fix the first issue.

rusefillc, 2021-04-13:
Fresh logs with CSV improvement

[logs.zip](https://github.com/rusefi/rusefi/files/6305380/logs.zip)

rusefillc, 2021-04-20:
i believe that the workaround has not solved the problem completely

[april20-with-workaround.zip](https://github.com/rusefi/rusefi/files/6340589/april20-with-workaround.zip)

andreika-git, 2021-04-28:
i believe that the faulty rusEFI_trigger_log_2021-04-20_00_23_11_956.logicdata was recorded WITHOUT the workaround, on an older firmware version.

rusefillc, 2021-04-28:
@andreika-git do you mean console without workaround? arm firmware was not changed right?

andreika-git, 2021-04-29:
Yes. It seems that the console used to record this file didn't have the workaround (AFAIK the workaround only changes CompositeParser.java, not the firmware).

How could I know it? 
* If I take the LogicdataStreamFileCsvSandbox converter WITHOUT the workaround and convert rusEFI_trigger_log_2021-04-20_00_23_11_958.csv to .logicdata, I get the result almost byte-identical to yours (i.e. the faulty file).
* But if a take the same java converter WITH the workaround, it produces a completely different file (larger by 20 bytes), and the resulting .logicdata is recognized by the viewer!

rusefillc, 2022-11-27:
Looks like files with less than 32768 events are fine while files with more than 32768 events are failing

Looks like some magic IDs are version-specific. @andreika-git do you remember what _exact_ version of Logic Analyzer software you have reverse-engineered?

rusefillc, 2023-01-11:
1.2.40 allows 16 channel simulation, actually 16D+16A

[image]

[16-channel-digital-simulated.zip](https://github.com/rusefi/rusefi/files/10388628/16-channel-digital-simulated.zip)

rusefillc, 2023-01-29:
Related to #2963

closed 2023-01-29 by rusefillc

rusefillc, 2023-01-29:
On the positive side there is progress I can open file using 1.2.40 "open capture" action in UI

[image]