rusefi/rusefi #1571: Fuel level sensing should support a curve, not just linear

mck1117, 2020-07-05:
My fuel level sender is quite nonlinear:

[image]

Using linear interpolation between the endpoints, an indicated 80% is actually only 56% full.

OrchardPerformance, 2020-07-13:
Most level senders will be linear for the actual sender but suffer huge non-linearity in output as the tank is not a constant cross section. 
+1 for using a curve for fuel sender.

mck1117, 2020-07-13:
Yes, I think my sender is linear with respect to angle, but the angle isn't linear with respect to fuel level (my tank is however roughly a rectangular prism, so the level is roughly proportional to the fill).

closed 2021-03-19 by mck1117