rusefi/rusefi #1575: DFU file name inside default bundle

rusefi, 2020-07-06:
we have rusefi.bin but rusefi_.dfu (With "_")

[image]

@chuckwagoncomputing

rusefi, 2020-07-06:
Let's make it rusefi_default.dfu to make naming a bit more consistent?

"rusefi.bin" having same name across all bundles is another issue for another day

in light of recent performance optimizations we can actually get rid of "_no_assert" and have only one set of files in the default bundle, that would make things just a bit more consistent

@chuckwagoncomputing

chuckwagoncomputing, 2020-07-06:
It should have been rusefi.dfu, but it seems I introduced a bug. (No surprise)
If we rename it to rusefi_default, do you want the .bin changed too?

rusefi, 2020-07-06:
not bin yet please

closed 2020-07-07 by rusefi