rusefi/rusefi #1580: Log upload fails

stefanst, 2020-07-08:
Trouble with log upload. 
I can select it, it gets analyzed, I get "seems OK". After clicking upload it's nowhere to be found.
[2020-07-07_20.51.26.zip](https://github.com/rusefi/rusefi/files/4887901/2020-07-07_20.51.26.zip)

rusefi, 2020-07-08:
should be rusefi/web_backend issue

could be same root issue as https://github.com/rusefi/web_backend/issues/54

rusefi, 2020-07-08:
@andreika-git

rusefi, 2020-07-08:
@stefanst we can recover this log if you post your .ini file here

stefanst, 2020-07-08:
99% certain that's the correct *.ini
[rusefi_mre_f4.zip](https://github.com/rusefi/rusefi/files/4891987/rusefi_mre_f4.zip)

stefanst, 2020-07-08:
Screenshot of upload attempt
[image]

rusefi, 2020-07-08:
@andreika-git see report above - even with .ini uploaded we have an error?

stefanst, 2020-07-08:
There was no change in behavior observable after the new .ini was uploaded.

rusefi, 2020-07-08:
@stefanst please confirm that you have hit "upload"?

stefanst, 2020-07-08:
Upload has been hit. 
@andreika-git made changes and issue is resolved.

closed 2020-07-08 by stefanst