rusefi/rusefi #1581: migrate amazing idle from FSIO to less opaque TS

rusefi, 2020-07-08:
@stefanst FYI

rusefi, 2020-07-08:
@stefanst first step new "extra" properties

[image]

rusefi, 2020-07-08:
FSIO can access those via

cfg_ACIDLEEXTRAOFFSET
cfg_ACIDLEEXTRAMIN

rusefi, 2020-07-08:
Just switched FSIO into "extra" not "replace" logic

my testing uses the following formulas:
```
	engineConfiguration->useFSIO12ForIdleOffset = true;
	setFsioExpression(QUOTE(MAGIC_OFFSET_FOR_IDLE_OFFSET), "ac_on_switch 0 cfg_acIdleExtraOffset if" PASS_ENGINE_PARAMETER_SUFFIX);

	engineConfiguration->useFSIO13ForIdleMinValue = true;
	setFsioExpression(QUOTE(MAGIC_OFFSET_FOR_IDLE_MIN_VALUE), "ac_on_switch 0 cfg_acIdleExtraMin if" PASS_ENGINE_PARAMETER_SUFFIX);
```

stefanst, 2020-07-09:
This seems to work with following FSIO formulas:
1: "ac_on_switch rpm 900 > time_since_ac_on_switch 0.6 > & &" for AC compressor out
2: "ac_on_switch rpm 900 > time_since_ac_on_switch 0.3 > & &" for AC fan out
12: "ac_on_switch cfg_ACIDLEEXTRAOFFSET 0 if" for extra offset
13: "ac_on_switch cfg_ACIDLEEXTRAMIN 0 if" for extra min

Old values "offset 2" and "min 2" should be removed from TS

rusefillc, 2020-09-06:
@stefanst says

AC fan and AC compressor get switched using FSIO. Look at this tune: https://rusefi.com/online/view.php?msq=172
Under "FSIO Formulas" outputs 1 and 2 are AC compressor and AC fan respectively.

rusefillc, 2020-09-06:
We have acFanRelay and acRelayPin we just have to add control logic

mck1117, 2021-06-27:
done with #2745

closed 2021-06-27 by mck1117