rusefi/rusefi #1587: kinetis auto gen does not auto gen

rusefi, 2020-07-10:
kinetis build fails due to not updated generated stuff

and gen_config does not fail :(

rusefi, 2020-07-10:
@chuckwagoncomputing please please?

chuckwagoncomputing, 2020-07-11:
And even worse, every build configures for all boards. Which means if one fails, they all fail.
Why does kinetis have it's own gen_config script?

chuckwagoncomputing, 2020-07-11:
As @mck1117 has pointed out, there are a lot of orthogonality issues in the build scripts. That can be fixed, but there will be >=one more Jenkins-only script created. That means we can make changes to builds and break Jenkins without realizing it, but I think the risk is worth it, and I'll definitely do what I can to minimize the risk.

rusefi, 2020-07-11:
@chuckwagoncomputing kinetis has it's own script because it has it's own hw enums, i.e. some of the input parameters are custom for kinetis.

mck1117, 2020-07-11:
kinetis builds with the normal makefile just like stm32... Just look at the old `build-firmware.yaml` (prior to it generating bundle)

rusefi, 2020-07-11:
@mck1117 kinetis code generation is a bit custom

chuckwagoncomputing, 2020-07-11:
I've submitted a pull request for this. I'll open another issue for orthogonality issues.

PR #1589 (2020-07-11): Fix kinetis
This should fix #1587
closed 2020-07-11 by rusefi